Xilinx/RapidWright #669: Failed to reliably download file

jgoeders, 2023-05-05:
When running RapidWright from Python, as part of our Github actions CI, we occasionally (actually fairly often), run into an exception indicating a failure to download part files:

```
java.lang.RuntimeException: ERROR: Failed to reliably download file: data/devices/artix7/xc7a200t_db.dat
```

```
java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: ERROR: Failed to reliably download file: data/unisim_data.dat
```

We would like to force download of these files as part of our CI environment setup, rather than an-hoc while our scripts are running.  Is it possible to do this?

We plan to try and catch these exceptions and retry the download, but just wondering if you have anything behind the scenes that is doing this already...

Thanks!

clavin-xlnx, 2023-05-05:
Thank you for bringing this up!  I would like to better understand the failure mechanism so that we can make RapidWright more robust--I see this failure showing up in the CI log: https://github.com/byuccl/bfasst/actions/runs/4887204926/jobs/8725501108?pr=173, I might try to diagnose this.  This seems to probably stem from hitting some limitation of GitHub Action runners.  RapidWright also uses GitHub Actions for CI and we don't run into this issue.  Do you know if you are running out of disk space?

I've added #670 that prints out the URL that it was attempting to download from upon ERROR.  

We do have some documentation around data file handling here: http://www.rapidwright.io/docs/RapidWright_Data_Files.html, although, I think you are wanting to get the files through an alternate method until we resolve why RapidWright can download them directly.  

In the past, RapidWright would distribute larger downloads of zipped files that had everything needed to run any device.  However, downloading a couple gigabytes of files with each install or major update was cumbersome, so we switched to an on-demand download approach.  These files are still available for download with each release, but will be more than what you need for CI.  You could try using the caching mechanism of CI to mitigate this also.  

Let me dig a deeper to see if we can root cause the issue.

jgoeders, 2023-05-05:
Thanks for your help. 

>This seems to probably stem from hitting some limitation of GitHub Action runners. RapidWright also uses GitHub Actions for CI and we don't run into this issue. 

I apologize, I forgot to mention a critical detail: these are running on local runners (ie local machines at BYU).  This of course complicates the issue quite a bit as there's a good chance that is the cause of the issue.  It's possible it's a BYU network issue or an underlying bug in the local runner infrastructure.  

It may be challenging to fully debug this, but it would be great to be able to catch it and work around it.

> Do you know if you are running out of disk space?
Do, I don't think this is happening.  Also, I should have mentioned that the issue is non-deterministic.  When we run into the issue I can usually restart the github action and have it complete successfully.

clavin-xlnx, 2023-05-05:
> I apologize, I forgot to mention a critical detail: these are running on local runners (ie local machines at BYU). This of course complicates the issue quite a bit as there's a good chance that is [...]

I updated #670 with an alternate download method attempt after the native Java one fails (assuming that it is just failing and not downloading a partial file and then giving up).  You might try to see if that eliminates the behavior.  It uses the built-in Jython urllib library to download the file (slow, not ideal, but has been fairly robust in the past).  Let me know if that causes the issue to go away.

jgoeders, 2023-05-18:
@clavin-xlnx I managed to track down the issue.  Not sure why I didn't think about this earlier, but the issue is because our CI script runs multiple jobs in parallel.  So what's happening is multiple invocations of `Design.readCheckpoint` are being run simultaneously, and then there is collision in downloading the necessary files.  Here is how I reproduced the issue:

```python
import pathlib
import shutil
from bfasst import jpype_jvm

import concurrent.futures

import jpype
import jpype.imports

# Assuming this script runs in the rapidwright top-level directory
RAPIDWRIGHT_DIR = pathlib.Path(__file__).parent.resolve()

DCP = <path to dcp>
EDF = <path to EDIF>

def worker():
    import jpype
    import jpype.imports

    jpype.startJVM(
        classpath=[
            str(RAPIDWRIGHT_DIR / "bin"),
            *(str(s) for s in (RAPIDWRIGHT_DIR / "jars").glob("*.jar")),
        ]
    )
    from com.xilinx.rapidwright.design import Design

    Design.readCheckpoint(DCP, EDF)

def main():
    n_procs = 20
    executor = concurrent.futures.ProcessPoolExecutor(max_workers=n_procs)

    futures = []
    for _ in range(n_procs):
        futures.append(executor.submit(worker))

    for future in concurrent.futures.as_completed(futures):
        print(future.result())

    shutil.rmtree(RAPIDWRIGHT_DIR / "data")

if __name__ == "__main__":
    main()
```

This should be pretty easy to get around by just running it once on environment setup to get all of the files before doing things in parallel.  

I'll let you close the issue if you like -- but it's worth noting that this is the only case where we've run into problems by running multiple rapidwright processes in parallel.

clavin-xlnx, 2023-05-18:
Thank you for taking the time to diagnose the issue and providing the example.  In hindsight, we have experienced this issue before and have a solution for it but it didn't occur to me from the error message that this was the underlying problem.  

I will update our documentation to more clearly highlight this potential pitfall as we currently don't have a locking mechanism across independent RapidWright processes other than the file system.  

To resolve the potential collision hazard, you'll want to enumerate all the potential devices that you might want to have available prior to running any parallel runs.  The pre-requisite API to call prior to any parallel runs is `FileTools.ensureDataFilesAreStaticInstallFriendly(String... devices)`.  To run this from the command line for the `xc7a100t` and `xc7a200t` devices, you could run:

```bash
java com.xilinx.rapidwright.util.RapidWright -c 'from com.xilinx.rapidwright.util import FileTools; FileTools.ensureDataFilesAreStaticInstallFriendly("xc7a100t", "xc7a200t")'
```

This will not only download all necessary RapidWright files and device files for those mentioned, but it will also generate device cache files which could also be another potential collision hazard in parallel runs.  

Thanks again for providing closure on this issue.  I will close it once I update the documentation and I'll drop a link here.

jgoeders, 2023-05-22:
Thanks! This function worked great and seems to have solved our problem.

clavin-xlnx, 2023-06-08:
With the 2023.1.0 release, we've introduced [a new convenience wrapper for RapidWright](http://www.rapidwright.io/docs/Install.html#rapidwright-wrapper) which allows for different use scenarios.  [When properly added to your `PATH`](http://www.rapidwright.io/docs/Install.html#install-steps), it reduces the command above to:
```
rapidwright jython -c 'FileTools.ensureDataFilesAreStaticInstallFriendly("xc7a100t", "xc7a200t")'
```
Which avoids the need to set the `CLASSPATH` variable and automatically generates the Python/Jython import statement for RapidWright classes.

We've also augmented the documentation to better highlight the pitfalls we've experienced in this PR:
http://www.rapidwright.io/docs/RapidWright_Data_Files.html#avoiding-on-demand-download-generation-of-data-files.

closed 2023-06-08 by clavin-xlnx