Xilinx/RapidWright #697: [Maven] rapidwright as Maven dependency fails

eddieh-xlnx, 2023-06-07:
With a simplified `build.gradle`:
```
plugins {
    id 'java-library'
}

dependencies {
    api 'com.xilinx.rapidwright:rapidwright:2023.1.0'
}

repositories {
    mavenCentral()
}

sourceSets {
    main {
        java {
            srcDirs = ['src']
        }
    }
}
```

produces:
```
$ ./gradlew compileJava
> Task :compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Could not resolve all files for configuration ':compileClasspath'.
   > Could not resolve com.xilinx.rapidwright:rapidwright:2023.1.0.
     Required by:
         project :
      > Could not resolve com.xilinx.rapidwright:rapidwright:2023.1.0.
         > Could not parse module metadata https://repo.maven.apache.org/maven2/com/xilinx/rapidwright/rapidwright/2023.1.0/rapidwright-2023.1.0.module
            > missing 'group' at /variants[2]/capabilities[0]

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
1 actionable task: 1 executed
```

where it does look like the `group` field is empty for `variant[2]`'s `capabilities`:
```
$ curl https://repo.maven.apache.org/maven2/com/xilinx/rapidwright/rapidwright/2023.1.0/rapidwright-2023.1.0.module 
{
  <<SNIP>>
  "variants": [
    {
      <<SNIP>>
    },
    {
      <<SNIP>>
    },
    {
      "name": "testFixturesApiElements",
      <<SNIP>>
      "capabilities": [
        {
          "group": "",        // <<<<< *** EMPTY ***
          "name": "rapidwright-test-fixtures",
          "version": "unspecified"
        }
      ]
    },
    {
      <<SNIP>>
    }
  ]
}
```

eddieh-xlnx, 2023-06-07:
I should mention that changing the dependencies to:
```
api 'com.xilinx.rapidwright:rapidwright-api-lib:2023.1.0'
```
causes it to advance just fine.

clavin-xlnx, 2023-06-07:
It appears that the test-fixtures item is being included by default and it is missing some annotation.  We probably need to remove them or figure out how to populate the group/version field accordingly.

clavin-xlnx, 2023-06-07:
Relevant messages from `gradle`:

```
$ ./gradlew publishToMavenLocal --warning-mode all
Maven publication 'mavenJava' pom metadata warnings (silence with 'suppressPomMetadataWarningsFor(variant)'):
  - Variant testFixturesApiElements:
      - Declares capability com.xilinx.rapidwright:rapidwright-test-fixtures:unspecified which cannot be mapped to Maven
  - Variant testFixturesRuntimeElements:
      - Declares capability com.xilinx.rapidwright:rapidwright-test-fixtures:unspecified which cannot be mapped to Maven
These issues indicate information that is lost in the published 'pom' metadata file, which may be an issue if the published library is consumed by an old Gradle version or Apache Maven.
The 'module' metadata file, which is used by Gradle 6+ is not affected.

```

clavin-xlnx, 2023-06-07:
#698 should resolve this issue.  To fix, it is possible to publish the jars locally:
```
./gradlew publishToMavenLocal
```
And then in the other project, augment the `repositories` section as so:
```
repositories {
    mavenLocal() // Added
    mavenCentral()
}
```

eddieh-xlnx, 2023-06-07:
> #698 should resolve this issue.

Confirmed! It does indeed.

closed 2023-06-07 by clavin-xlnx