Xilinx/RapidWright #701: RWRoute only assigns a single SitePinInst to the ground net.

haydenc-amd, 2023-06-12:
I have an unrouted design checkpoint (shared below). When I attempted to route using RWRoute  (`rapidwright RWRoute cntr-preRouted.dcp cntr-routed.dcp --nonTimingDriven`), it only routes the GND net to a single site pin leaving the rest of the net unrouted.

[cntr-preRouted.zip](https://github.com/Xilinx/RapidWright/files/11710549/cntr-preRouted.zip)

eddieh-xlnx, 2023-06-13:
Hi @haydenc-amd. I'm not able to reproduce what you're seeing. This is my log on the latest commit (74ed97f):
```
$ rapidwright RWRoute cntr-preRouted.dcp cntr-routed.dcp --nonTimingDriven
==============================================================================
==                                 RWRoute                                  ==
==============================================================================
==============================================================================
==                     Reading DCP: cntr-preRouted.dcp                      ==
==============================================================================
 XML Parse & Device Load:     0.927s
      (Read XDEF Header):            (    0.055s)
   (Read XDEF Placement):            (    0.084s)
            (EDIF Parse):            (    0.444s)
     (Read XDEF Routing):            (    0.232s)
     Read Netlist & XDEF:     0.546s
               Hier Info:     0.033s
------------------------------------------------------------------------------
         [No GC] *Total*:     1.507s
WARNING: Not masking nodes across RCLK could result in delay optimism.
==============================================================================
==                               Route Design                               ==
==============================================================================
INFO: Routing 31 pins of GLOBAL_LOGIC0
INFO: Routing 1 pins of GLOBAL_LOGIC1
------------------------------------------------------------------------------
              Generated       RRG        Routed  Nodes With          Total Run
Iteration     RRG Nodes  Time (s)   Connections    Overlaps           Time (s)
---------  ----------------------   -----------  ----------   ----------------
   1               3823      0.01            64          21               0.04
   2                429      0.00            33           8               0.01
   3                352      0.00            11           2               0.00
   4                102      0.00             2           0               0.00
------------------------------------------------------------------------------

INFO: Route 3 direct connections 

INFO: No PIP overlaps

==============================================================================
==                                Statistics                                ==
==============================================================================
Total wirelength:                           41
Route design:                            0.37s
├─ Initialization:                       0.11s
└─ Routing:                              0.26s
==============================================================================
     (Build XDEF Header):            (    0.003s)
            (Write EDIF):            (    0.052s)
(Write XDEF Cache (2/3)):            (    0.041s)
(Write XDEF Cache (1/3)):            (    0.042s)
(Write XDEF Cache (3/3)):            (    0.047s)
  (Write XDEF Placement):            (    0.078s)
    (Write XDEF Routing):            (    0.867s)
 (Write XDEF Finalizing):            (    0.004s)
            (Write XDEF):            (    0.896s)
       Write EDIF & XDEF:     0.904s
               Write XDC:     0.005s
               Hier Info:     0.010s
------------------------------------------------------------------------------
         [No GC] *Total*:     0.919s

INFO: Write routed design
 cntr-routed.dcp
```

Loading the routed design into Vivado and doing `report_route_status` also comes back clean.

Are you seeing something different?

haydenc-amd, 2023-06-13:
@eddieh-xlnx when I run `rapidwright RWRoute cntr-preRouted.dcp cntr-routed.dcp --nonTimingDriven`, I don't see the following lines:
```
INFO: Routing 31 pins of GLOBAL_LOGIC0
INFO: Routing 1 pins of GLOBAL_LOGIC1
```

Additionally, when I load the routed design in Vivado and run `report_route_status` I get:
```
Design Route Status
                                               :      # nets :
   ------------------------------------------- : ----------- :
   # of logical nets.......................... :         104 :
       # of nets not needing routing.......... :          67 :
           # of internally routed nets........ :          64 :
       # of routable nets..................... :          37 :
           # of unrouted nets................. :           2 :
           # of fully routed nets............. :          35 :
       # of nets with routing errors.......... :           3 :
           # of nets with no driver........... :           3 :
           # of nets with some unrouted pins.. :           3 :
   ------------------------------------------- : ----------- :
```
Running `get_nets -filter {ROUTE_STATUS == "UNROUTED"}` shows that \<const0\> and \<const1\> are the two unrouted nets.

I made sure to pull from master and ran `./gradlew compileJava` before running these tests today.

haydenc-amd, 2023-06-13:
@eddieh-xlnx I accidentally uploaded a version of the counter that didn't have any flip flops. I don't know why, but when the checkpoint doesn't have FFs, it seems to route GND and VCC just fine. Here is the checkpoint with FFs. Sorry for the troubles!

[cntr-preRoutedFFs.zip](https://github.com/Xilinx/RapidWright/files/11739222/cntr-preRoutedFFs.zip)

PR #703 (2023-06-14): Fix DesignTools.makePhysNetNamesConsistent() 
By transforming logical static nets named `<const{0,1}>` into physical static nets `GLOBAL_LOGIC{0,1}`

Fixes #701.

CC: @haydenc-amd
closed 2023-06-14 by eddieh-xlnx