Xilinx/RapidWright #709: Problem with Cell.getCompatiblePlacements() - hdi_primitives library not recognized as part of design netlist

reillymck, 2023-06-15:
I'm trying to get the default l2p mappings for a UNISIM cell. From what I understand, I need to place the cell on an empty site, and then query Cell.getPinMappingsL2P(). Anyway, when I try to query compatible placement options for the Unisim cell, I'm getting an error that my netlist doesn't have the hdi_primitives library. But when I try to add that netlist to my design's netlist library, it says that it already exists. Maybe I am missing something here?
```
>>> from com.xilinx.rapidwright.design import Design, Unisim
>>> design = Design.readCheckpoint(("design.dcp")
>>> c4 = design.createCell("test_c4", Unisim.CARRY4)
>>> options = c4.getCompatiblePlacements()

Traceback (most recent call last):
  File "<console>", line 1, in <module>
java.lang.java.lang.RuntimeException:  ERROR: Please add the following library to your design's netlist: hdi_primitives

>>> prim_lib = design.getUnisimCell(Unisim.CARRY4).getLibrary()
>>> prim_lib.getName()
'hdi_primitives'
>>> netlist = design.getNetlist()
>>> netlist.getName()
'natalius_8bit_risc'
>>> netlist.addLibrary(prim_lib)

Traceback (most recent call last):
  File "<console>", line 1, in <module>
java.lang.java.lang.RuntimeException: ERROR: EDIFNetlist already has library named hdi_primitives
```
Really, at the end of the day, I hope to temporarily place a UNISIM cell, and then be able to get the default pin mappings, like for a CARRY4:
```
"CARRY4": dict(
            {
                "DI[3]": "DI3",
                "DI[2]": "DI2",
                "DI[1]": "DI1",
                "DI[0]": "DI0",
                "CI": "CIN",
                "CO[3]": "CO3",
                "CO[2]": "CO2",
                "CO[1]": "CO1",
                "CO[0]": "CO0",
                "O[3]": "O3",
                "O[2]": "O2",
                "O[1]": "O1",
                "O[0]": "O0",
                "S[3]": "S3",
                "S[2]": "S2",
                "S[1]": "S1",
                "S[0]": "S0",
                "CYINIT": "CYINIT",
            }
        ),
``` 
If there is a better way to do this, I am also interested in that. 
I have attached my design.dcp for debugging help (renamed to [design.zip](https://github.com/Xilinx/RapidWright/files/11762440/design.zip))

clavin-xlnx, 2023-06-20:
Sorry for the delay, we are working on a fix for this issue, will update you soon.

clavin-xlnx, 2023-06-22:
@reillymck - the 2023.1.1 release should fix the issue you mentioned, would you mind giving it a try?

reillymck, 2023-06-27:
Yes I will try it out. Sorry I missed the notification for your comment.

reillymck, 2023-06-27:
So I checked out the 2023.1.1 release:
```
git checkout tags/v2023.1.1-beta
./gradlew updateJars
```
Then I started the jvm with my rapidwright classpath. However, when I try to read a checkpoint, I'm getting the following error:
```
>>> from com.xilinx.rapidwright.design import Design
>>> design = Design.readCheckpoint("/home/reilly/equiv/bfasst/design.dcp")
==============================================================================
==                         Reading DCP: design.dcp                          ==
==============================================================================
Traceback (most recent call last):
  File "com.xilinx.rapidwright.design.Design.java", line -1, in com.xilinx.rapidwright.design.Design.readCheckpoint
Exception: Java Exception

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
java.lang.java.lang.NoSuchMethodError: java.lang.NoSuchMethodError: 'com.esotericsoftware.kryo.io.Input com.xilinx.rapidwright.util.FileTools.getKryoZstdInputStream(java.io.InputStream)'
```
Did I do something wrong here?

clavin-xlnx, 2023-06-27:
> Did I do something wrong here?

This may be obvious, but did you compile the code?
```
./gradlew compileJava
```

reillymck, 2023-06-28:
That was it... awkward. Anyway, Cell.getCompatiblePlacements() seems to be working now. Thanks for the prompt fix!

reillymck, 2023-06-28:
Sorry, not sure if you wanted to close it when the beta branch was marked as a release. Feel free to re-open.

closed 2023-06-28 by reillymck

clavin-xlnx, 2023-06-29:
> That was it... awkward. Anyway, Cell.getCompatiblePlacements() seems to be working now. Thanks for the prompt fix!

I'm updating the directions on the README.md https://github.com/Xilinx/RapidWright/pull/733/files?short_path=b335630#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5 so that it is more explicit.   

Thanks for filing the issue.