Xilinx/RapidWright #742: Unable to use rapidwright.bitstream APIs on python

yongsk, 2023-07-10:
I was able to import most of the packages on python except rapidwright.bitstream with AttributeError saying it has no attribute 'bitstream'.

Bitstream APIs are not provided to python?
How can I use them?

Thank you.

clavin-xlnx, 2023-07-10:
You should be able to import any of the Java classes found in RapidWright in Python.  Here is an example:

Quickly setup RapidWright in a Python virtual environment:
```
mkdir bitstream_example
cd bitstream_example
python3 -mvenv env
source env/bin/activate
pip3 install --upgrade pip
pip3 install rapidwright
```
You can either run the following in a script or in the interpreter:
```
import rapidwright
from com.xilinx.rapidwright.bitstream import Bitstream
from com.xilinx.rapidwright.device import Device   
b = Bitstream("emptyExampleBitstream", Device.AWS_F1)
b.writeBitstream("empty.bit")
```

If you are still having trouble, make sure you are using the latest RapidWright and then please respond back with your environment, and error messages.

dg-deus, 2023-08-07:
I still have the same problem when importing bitstream API with the code below.
`jpype.startJVM(classpath=["rapidwrite-2023.1.2-standalone-win64.jar"])`

However, I was able to import the bitstream API in the exactly same environment with old version, 2021.2.0.
Something changed after the 2021.2.0 version?

clavin-xlnx, 2023-08-07:
> I still have the same problem when importing bitstream API with the code below. `jpype.startJVM(classpath=["rapidwrite-2023.1.2-standalone-win64.jar"])`

The standalone jar name appears to be mispelled, the correct filename is:
```
jpype.startJVM(classpath=["rapidwright-2023.1.2-standalone-win64.jar"])
```

dg-deus, 2023-08-11:
> The standalone jar name appears to be mispelled, the correct filename is:
>
> ```
> jpype.startJVM(classpath=["rapidwright-2023.1.2-standalone-win64.jar"])
> ```

You were right, that was typo. After I got it fixed and tried again, it still falied to import bitstream API.

clavin-xlnx, 2023-08-15:
I was able to reproduce the issue.  It turns out in 2022.2.2 we changed the way we zipped up the standalone jars to use `zipmerge` and for some reason, this change is preventing the `Bitstream` class file from being found in the archive jar.  I've re-generated the 2023.1.2 standalone jars and it appears to have resolved the issue.

Please try re-downloading the latest jar: 

```
curl -L -O https://github.com/Xilinx/RapidWright/releases/download/v2023.1.2-beta/rapidwright-2023.1.2-standalone-win64.jar
```

and try again.

eddieh-xlnx, 2023-10-25:
@dg-deus Can you confirm if this issue is resolved for you too?

closed 2024-01-17 by eddieh-xlnx