Xilinx/RapidWright #786: getByteOffset method in Frame class

dg-deus, 2023-08-17:
I see "getByteOffset" method in rapidwright/javadoc, but I cannot find it when using RapidWright with python.

Is this not implemented yet, or is there something wrong with my environments?

Thank you in advance.

clavin-xlnx, 2023-08-17:
Sorry, the getByteOffset entry in the Javadoc is an error, we currently don't have a API for that.  Are you able to share more about what you are trying to do?  Perhaps there is another way we can help.

dg-deus, 2023-09-12:
In some cases, I want to know where some LUTs reside in bitstream.
I thought "getByteOffset" might help me to find the solution.

clavin-xlnx, 2023-10-27:
Sorry for the delayed response.  The getByteOffset entry has been removed from the javadoc.  

As for locating the LUTs in the bitstream, we can only provide information about where SRL and LUTRAM type primitives exist in the bitstream as that is documented in the [Logic Location file](https://docs.xilinx.com/r/en-US/ug908-vivado-programming-debugging/Changing-the-Bitstream-File-Format-Settings) (`.ll`) provided from Vivado.  The current documentation on how to update those state bits related to LUTRAMs is here https://web.xilinx.com/~clavin/RapidWrightPages/docs/Bitstream_Manipulation.html.  However, explicit locations can be found in a design by using the Logic Location file.

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx