Xilinx/RapidWright #789: Interchange writer misses belPin

zakn-amd, 2023-08-21:
from commit `f0a8687ee0eda772725e90c80259afff3b5b1569`
Attached physical netlist (with .txt extension so that github doesn't reject it) was generated as follows:
```
rapidwright DcpToInterchange picoblaze_ooc_X10Y235.dcp # from rapidwright DCPs
```
[picoblaze_ooc_X10Y235.phys.txt](https://github.com/Xilinx/RapidWright/files/12401678/picoblaze_ooc_X10Y235.phys.txt)

The net `processor/alu_decode1_lut/O6` seems to be missing the belPin AX on the CARRY8 in SLICE_X15Y238. 

A completely nonstandard printout of the physNet (output of a recursive DF traversal):
```
belPin SLICE_X16Y239 C6LUT O6 {                                                 
 sitePin SLICE_X16Y239 C_O {                                                    
  pip INT_X11Y239 LOGIC_OUTS_W4 INT_NODE_SDQ_51_INT_OUT1 False False {          
   pip INT_X11Y239 INT_NODE_SDQ_51_INT_OUT1 WW4_W_BEG1 False False {            
    pip INT_X9Y239 WW4_W_END1 INT_NODE_SDQ_51_INT_OUT0 False False {             
     pip INT_X9Y239 INT_NODE_SDQ_51_INT_OUT0 EE2_W_BEG1 False False {            
      pip INT_X10Y239 EE2_W_END1 INT_NODE_SDQ_49_INT_OUT1 False False {         
       pip INT_X10Y239 INT_NODE_SDQ_49_INT_OUT1 SS1_W_BEG1 False False {        
        pip INT_X10Y238 SS1_W_END1 INT_NODE_SDQ_50_INT_OUT0 False False {       
         pip INT_X10Y238 INT_NODE_SDQ_50_INT_OUT0 INT_INT_SDQ_60_INT_OUT1 False False {
          pip INT_X10Y238 INT_INT_SDQ_60_INT_OUT1 INODE_E_9_FT1 False False {   
           pip INT_X10Y238 INODE_E_9_FT1 BOUNCE_E_0_FT1 False False {            
            sitePin SLICE_X15Y238 AX                                            
 }}}}}}}}}}}                                                                    
 {belPin SLICE_X16Y239 C_O C_O}                                                 
}
```
shows that the long branch (that goes through a bunch of intersite routing terminates on a sitePin, instead of a belPin)

eddieh-xlnx, 2023-08-22:
So it appears there are two issues here.
1. At the source end it goes BELPin -> SitePin -> BELPin, which is incorrect. This is fixed in #790.
2. At the sink end, the net terminates at a SitePin. I would expect it be followed by at least two more BELPins (site port and then actual leaf pin)

Looking at 2 some more it appears that the `processor/alu_decode1_lut/O6` net (output of LUT6) has another alias (`processor/arith_carry_in`, input to CARRY8) :
[image]
and it turns out that both logical nets appear as physical nets, with the `arith_carry_in` net attached to the CARRY8's site wire.

Since both physical nets exist, PhysNetWriter doesn't put them together as a single net.

The heavy-handed solution would be to call `DesignTools.makePhysNetNamesConsistent()` to bring those physical nets together, but that could be an expensive operation (especially the EDIFNetlist.generateParentNetMap() dependency) to call before every PhysNet write.

What's interesting here is that the `LUT6` is the result of a `LUT6_2` macro expansion. Would it make sense to only do a lightweight `makePhysNetNamesConsistent()` on those nets that are affected?

eddieh-xlnx, 2023-08-22:
Test for 2 here: https://github.com/Xilinx/RapidWright/compare/bel_then_site_pin...fix_789?expand=1

clavin-xlnx, 2023-08-22:
> What's interesting here is that the `LUT6` is the result of a `LUT6_2` macro expansion. Would it make sense to only do a lightweight `makePhysNetNamesConsistent()` on those nets that are affected?

Do we believe that the duplicate physical nets with different aliases are caused by macro expansion?  

Are we able to efficiently detect duplicate physical nets with different alias names?

Should we revisit the implementation of `makePhysNetNamesConsistent()` to make it more efficient?

eddieh-xlnx, 2023-08-22:
> Do we believe that the duplicate physical nets with different aliases are caused by macro expansion?

Yes: without the `LUT6_2` macro being expanded, the parent net of its output is the one that feeds into the `CI` pin. It's only after macro expansion that the parent net moves "upstream" to the net coming from `LUT6.O`.

> Are we able to efficiently detect duplicate physical nets with different alias names?

Detection would be to query `EDIFNetlist.getParentNetName()` of every net encountered by PhysNetlistWriter; or to use `DesignTools.makePhysNetNamesConsistent()`.

> Should we revisit the implementation of makePhysNetNamesConsistent() to make it more efficient?

I think the bulk of the runtime in `makePhysNetNamesConsistent()` is generating the parent net map, assuming it hasn't been done already. I think are speedup opportunities there, but I'm hoping to find an easier way to resolve this issue for written PhysicalNetlists since it alone has no concept of aliases.

clavin-xlnx, 2023-08-22:
> Sorry, let me explain again. I think this is an artifact of macro expansion -- whether that happens in Vivado or RapidWright. Although Vivado accepts and can handle net aliases as sitewires, RapidWright can't. My thought was that perhaps there can be a lightweight `makePhysNetNamesConsistent()` that only examines expanded macros, and does so without having to build a parent net map because it knows what's inside the macro is a parent net, and what's outside of the macro must be an alias.

I've been looking at this kind of scenario in a few other DCPs, I've only seen the non-parent alias used in the picoblaze design.  My hunch is that this inconsistent alias issue is rare and that the macro expansion is not necessarily related.  Do we have another design that exhibits this behavior?  

I don't have objections to a light weight version of `makePhysNetNamesConsistent()`, but I'd like to seem more evidence that it happens around macro expansion beyond this single instance.

clavin-xlnx, 2023-08-22:
Another thought, if we do determine that this is rare and semi-unpredictable, could we throw a special exception and then catch higher up in the stack, and run `makePhysNetNamesConsistent()` before trying again?  That could give us some ability to support these rare cases but also avoid the overhead of making a call to `makePhysNetNamesConsistent()` for all designs.

eddieh-xlnx, 2023-08-22:
> I've been looking at this kind of scenario in a few other DCPs, I've only seen the non-parent alias used in the picoblaze design. My hunch is that this inconsistent alias issue is rare and that the macro expansion is not necessarily related. Do we have another design that exhibits this behavior?

That's reasonable: I haven't seen examined in detail when this happens elsewhere. For most of our RapidWright usage, we do call `makePhysNetNamesConsistent()` before calling RWRoute, etc. In the situation where all you want to do is to convert a DCP into a PhysicalNetlist, enforcing a `makePhysNetNamesConsistent()` by default seems a little excessive.

I am on board with the wait-and-see approach, as I still can't think of straightforward way of performing expanded-macro-only operations without walking the entire design.

PR #790 (2023-08-22): [PhysNetlistWriter] Insert site port BELPin before site pin
Partially fixes #789, from where testcase was derived.
closed 2023-08-22 by eddieh-xlnx