Xilinx/RapidWright #814: can't read "io_pin_sites": no such variable @ 030-iob18 fuzzer

dg-deus, 2023-09-11:
I keep having the error message,can't read "io_pin_sites": no such variable, while executing 030-iob18 fuzzer.

So, I looked into generate.tcl in 030-iob18 directory.
And I opened my Vivado 2017.2 and tested make_io_pin_sites proc after "link_design -part xc7a15t" command.
As a result, it seems like there are no site and pad with IOB18* site type, skipping `dict append io_pin_sites $site $pad`.
That's why I constantly have the error.
Does this 030-iob18 fuzzer depend on target FPGA part?
 
`proc make_io_pin_sites {} {
    # get all possible IOB pins
    foreach pad [get_package_pins -filter "IS_GENERAL_PURPOSE == 1"] {
        set site [get_sites -of_objects $pad]
        if {[llength $site] == 0} {
            continue
        }
        if [string match IOB18* [get_property SITE_TYPE $site]] {
            dict append io_pin_sites $site $pad
        }
    }
    return $io_pin_sites
}
`

clavin-xlnx, 2023-09-11:
I think this issue is not directly related to RapidWright.  Did this issue get filed for the wrong project?

dg-deus, 2023-09-12:
Sorry, I posted wrong project.
My apologies.

closed 2023-09-12 by dg-deus