Xilinx/RapidWright #826: NullPointerException at edif.EDIFHierPortInst.getPhysicalCell

nqdtan, 2023-09-19:
My code is as follows.

```java
public static void main(String[] args) throws FileNotFoundException {
    Design design = Design.readCheckpoint("socket_cc.dcp", "socket_cc.edf");
    EDIFNetlist netlist = design.getNetlist();

    for (Net net : design.getNets()) {
      EDIFHierNet ehnet = net.getLogicalHierNet();
      if (ehnet == null)
        continue;

      for (EDIFHierPortInst ehpi : ehnet.getLeafHierPortInsts()) {
        if (ehpi == null)
          continue;

        System.out.println("EDIFHierPortInst: " + ehpi);
        System.out.println("PhysicalCell: " + ehpi.getPhysicalCell(design));
      }
    }
```

I got the following error

>...
EDIFHierPortInst: design_1_i/socket/socket_top_0/inst/socket_template/controller/ff_lsu0_ram_seg_stride/buffer/mem_reg_0_15_0_13/RAMH_D1/ADR0
PhysicalCell: design_1_i/socket/socket_top_0/inst/socket_template/controller/ff_lsu0_ram_seg_stride/buffer/mem_reg_0_15_0_13/RAMH_D1(BEL: H6LUT)
EDIFHierPortInst: design_1_i/socket/socket_top_0/inst/imem/mem_reg_384_447_4_4/DP/RAMD64_INST/O
Exception in thread "main" java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.edif.EDIFNetlist.getTopCellInst()" because the return value of "com.xilinx.rapidwright.edif.EDIFLibrary.getNetlist()" is null
	at com.xilinx.rapidwright.edif.EDIFHierPortInst.getFullHierarchicalInstName(EDIFHierPortInst.java:91)
	at com.xilinx.rapidwright.edif.EDIFHierPortInst.getPhysicalCell(EDIFHierPortInst.java:203)
	at com.xilinx.rapidwright.examples.Test.main(Test.java:25)

**RapidWright version**: I have observed similar error with v2023.1.0 --> 2023.1.3

However, using v2022.2.3, I was able to retrieve the physical cell **design_1_i/socket/socket_top_0/inst/imem/mem_reg_384_447_4_4/DP/RAMD64_INST** correctly.

Please find the code + checkpoint files attached (checkpoint is created by Vivado 2022.2). Thank you!

[test.zip](https://github.com/Xilinx/RapidWright/files/12659923/test.zip)

clavin-xlnx, 2023-09-20:
Thanks for bringing this up!  I believe I have a fix for this issue in https://github.com/Xilinx/RapidWright/pull/828.  

The problem was that we made a change to deep copy cells (#672), but we didn't deep copy their children and add them to the netlist libraries so when `getPhysicalCell()` called out to get the top cell, it couldn't on these macro children.  This change mentioned above should resolve this by deep copying the child cell types if they are not already present.

nqdtan, 2023-09-20:
Works well for my case. Thanks!

closed 2023-09-20 by nqdtan