Xilinx/RapidWright #845: [WARNING: Hierarchy information reference to non-existent net XXX] Using ModuleInst

nqdtan, 2023-10-08:
I got the following warnings when reading a design checkpoint (attached) generated from another flow (that was created using ModuleInst)

>...WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/ff_bridge/net_b714
WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/ff_bridge/net_b713
WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/ff_bridge/net_b719
WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/ff_bridge/net_b718
WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/ff_bridge/net_b717
WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/clk_BUFG_LOPT_OOC
WARNING: Hierarchy information reference to non-existent net: socket_cl_inst/socket_reset

And the code is simply as follows

```java
Design design = Design.readCheckpoint("socket.dcp", "socket.edf");
Module module = new Module(design);
Design newDesign = new Design("new_top_design", design.getPartName());
ModuleInst modInst = newDesign.createModuleInst("socket_inst", module);
newDesign.writeCheckpoint("new_top_design.dcp");
```

I got the following error (besides the warnings above)

>Exception in thread "main" java.lang.NullPointerException: Cannot invoke "java.util.List.size()" because "<local8>" is null
	at com.xilinx.rapidwright.design.ModuleCache.a(Unknown Source)
	at com.xilinx.rapidwright.design.ModuleCache.writeDesignHierarchy(Unknown Source)
	at c.b(Unknown Source)
	at c.a(Unknown Source)
	at com.xilinx.rapidwright.design.Design.writeCheckpoint(Unknown Source)
	at com.xilinx.rapidwright.spades.Test2.main(Test2.java:70)
	Suppressed: java.io.IOException: This archive contains unclosed entries.
		at org.python.apache.commons.compress.archivers.zip.ZipArchiveOutputStream.finish(ZipArchiveOutputStream.java:467)
		at org.python.apache.commons.compress.archivers.zip.ZipArchiveOutputStream.close(ZipArchiveOutputStream.java:966)
		... 4 more

I'm certain that these warning messages do not affect my design functionality, since I have a different flow that works with it without any issues. For now, to work with `ModuleInst`, I have to remove the file `RW_HIER` from the input DCP so that RapidWright does not need to parse any hierarchy information..

I'd like to understand why such warnings happen, and possible ways to suppress/remove them so that I don't have to edit my DCP. (The code for generating the DCP is rather tedious, and may be too detailed than necessary, so I doubt you will need it. But let me know otherwise.)

Thanks!

[DCP](https://drive.google.com/file/d/1BQzmbtnDoeEeAewwdQIQYzTWhgUMChYv/view?usp=sharing)

clavin-xlnx, 2023-10-09:
The warnings are likely due to the module instances not being connected properly to the rest of the design.  When looking at the first warning, it appears that the referenced net, `socket_cl_inst/ff_bridge/net_b714` is not actually the parent physical net name:

[image]

Physical nets are typically named after their driver logical net, in this case it would be: 

```
socket_cc_inst/socket_top_0/inst/rg_cl_intf_pipe/lsu1_port1_d_pb/genblk1[0].pipe_reg/lsu1_port1_d[10]
```

As shown in the "Aliases" tab:
[image]

However, the name `socket_cl_inst/ff_bridge/net_b714` was referenced by the `ModuleInst` as a physical net.  This is technically incorrect given the connectivity, however, what might have happened is that `socket_cl_inst` was created separately from `socket_cc_inst` and then connected later.  One possible fix that might resolve this in the DCP generation of the example provided would be to run `DesignTools.makePhysNetNamesConsistent(Design)` on the design before writing it to a DCP.  However, this is more of a patch of an underlying problem which is that the two components were not properly connected in the first place.  If you are able to share that code and allow us to reproduce it, we can help identify the issue.

Regarding the error when writing the DCP, the simplest way to remove the generation of the `RW_HIER` file is to simply flatten the physical netlist hierarchy by calling `Design.flattenDesign()`.  This will remove all the `Module` and `ModuleInst` objects from the design and leave behind their flattened implementations.  Again, I believe that the reason for this error is that some internal state is not consistent and likely leading to the failure.

nqdtan, 2023-10-10:
Hi Chris,

It is indeed as you have guessed. The two designs, `socket_cc_inst` and `socket_cl_inst` are implemented separately in Vivado. I stitch them in RapidWright. The nets shown in the warnings indeed originate from outside `socket_cl_inst`. I tried `flattenDesign` as well as `DesignTools.makePhysNetNamesConsistent(Design)`, but they didn't seem to fix the issue (still got the warning messages and the error at the end). For now, I fixed the issue by removing those nets in `socket_cl`'s `Module` object (before instantiating `ModuleInst`). The warnings are gone, but the error at `writeCheckpoint` still persists.

> Exception in thread "main" java.lang.NullPointerException: Cannot invoke "java.util.List.size()" because "<local8>" is null
	at com.xilinx.rapidwright.design.ModuleCache.a(Unknown Source)
	at com.xilinx.rapidwright.design.ModuleCache.writeDesignHierarchy(Unknown Source)
	at c.b(Unknown Source)
	at c.a(Unknown Source)
	at com.xilinx.rapidwright.design.Design.writeCheckpoint(Unknown Source)
	at com.xilinx.rapidwright.spades.Test2.main(Test2.java:71)
	Suppressed: java.io.IOException: This archive contains unclosed entries.
		at org.python.apache.commons.compress.archivers.zip.ZipArchiveOutputStream.finish(ZipArchiveOutputStream.java:467)
		at org.python.apache.commons.compress.archivers.zip.ZipArchiveOutputStream.close(ZipArchiveOutputStream.java:966)

Please find the code + relevant checkpoint files for generating `socket.dcp` attached. I use additional methods that I defined in `EDIFNetlist.java` of my local RapidWright copy (check `edifnetlist.patch`). I'm sorry that the code could appear a bit unorganized. Let me know if you need any clarifications. In general, I extracted `socket_cc` from a child cell of `checkpoints/socket_cc.dcp`, and I used `ModuleInst` to extract `socket_cl` from the top level cell of `checkpoints/socket_rg_cl.dcp`. I would have liked to use `ModuleInst` to copy `socket_cc` as well, but encountered some issues, so I used my own methods for copying cells and net routings from the reference design. Another thing is that the two designs have some overlaps -- essentially a column of FFs (I called it ff_bridge). I used them to guide the stitching between the interfaces of the two designs (so that I don't have to perform any actual routing). Thank you for taking a look.

[socket_stitch](https://drive.google.com/file/d/1j39QqizYAzuK9f7L9Pl8zpLyLPiYMmt6/view?usp=sharing)

clavin-xlnx, 2023-10-10:
Hi Tan, 

I was unable to reproduce the error, but was able to successfully generate a DCP and load it into Vivado.  One thing I noticed is that there appears to be encrypted cells that were part of the design, but the `*.edn` files are missing.  Maybe you could double check.

There are a number of routing issues when I load the result in Vivado, but they could be due to the missing encrypted IP, here are some of the messages that indicate they are missing: 
[image]

One other way to move an implementation into a design is `DesignTools.copyImplementation()` which if two designs contain the same instance name cell and their ports match, you can copy the implementation from one design to another.  I'm not quite sure if it is what would work for your situation here, but it might be worth a look.

nqdtan, 2023-10-10:
Thanks Chris. Are you able to load the generated DCP in RapidWright just to write it out again (this is because I need to pass this DCP to another RapidWright flow.) -- as in my original test code?

```java
Design design = Design.readCheckpoint("socket.dcp", "socket.edf");
Module module = new Module(design);
Design newDesign = new Design("new_top_design", design.getPartName());
ModuleInst modInst = newDesign.createModuleInst("socket_inst", module);
newDesign.writeCheckpoint("new_top_design.dcp");
```

I hit the error as I mentioned in the previous post. My workaround is removing `RW_HIER` from the DCP, but it would be better if I don't have too. It must be something related to how I used `ModuleInst`.

I added the missing `*.edn` files in the attachment. The DCP can be fully constructed in Vivado by `source socket_load.tcl`

I'll check out `DesignTools.copyImplementation()`. Thanks for the advice.

[socket_stitch](https://drive.google.com/file/d/1FcYIj01oFDmy1ZMErUuQHFi9YLxKsE3G/view?usp=sharing)

nqdtan, 2023-10-10:
Ah, I found that if I add `newDesign.flattenDesign()`, it will work fine.

```java
Design design = Design.readCheckpoint("socket.dcp", "socket.edf");
Module module = new Module(design);
Design newDesign = new Design("new_top_design", design.getPartName());
ModuleInst modInst = newDesign.createModuleInst("socket_inst", module);
newDesign.flattenDesign();
newDesign.writeCheckpoint("new_top_design.dcp");
```

closed 2023-10-11 by nqdtan

clavin-xlnx, 2023-10-11:
Ok, let us know if you want to discuss this further.