Xilinx/RapidWright #849: Device data generate

the-centry, 2023-10-11:
How to generate the device database rather than download the data?
Thanks!

clavin-xlnx, 2023-10-11:
The RapidWright device database files are generated as part of an internal process.  It is currently not possible to do it externally.  Can you share what you would like to accomplish?  Perhaps we can help you in another way.

the-centry, 2023-10-14:
> The RapidWright device database files are generated as part of an internal process. It is currently not possible to do it externally. Can you share what you would like to accomplish? Perhaps we can help you in another way.

Year, I want to know the device generate rule and change some data to create a device!

clavin-xlnx, 2023-10-16:
> Year, I want to know the device generate rule and change some data to create a device!

Although the device model in RapidWright is part of the API library, another alternative would be to use the FPGA Interchange Format which is an open schema that enables an open source implementation of device models.  RapidWright fully supports the [FPGA Interchange Format](https://www.rapidwright.io/docs/FPGA_Interchange_Format.html) which has schemas to represent device models, and logical and physical netlists of partially or fully implemented designs.  In theory, you could export a device model from RapidWright in the FPGA Interchange Format device model and then modify it.

the-centry, 2023-10-17:
> Although the device model in RapidWright is part of the API library, another alternative would be to use the FPGA Interchange Format which is an open schema that enables an open source [...]

A good idea! Thanks so much!

closed 2023-10-17 by the-centry