Xilinx/RapidWright #853: DREAMPlaceFPGA tutorial missing the step to generate the device IF file

pwang7, 2023-10-15:
According to the [documentation](https://github.com/rachelselinar/DREAMPlaceFPGA/blob/main/IFsupport/README.md) of DREAMPlaceFPGA, before placing the design using DREAMPlaceFPGA, it has to generate the device IF file for `xcvu3p-ffvc1517-2-e` and then place it into the `IFsupport` directory of DREAMPlaceFPGA.

clavin-xlnx, 2023-10-16:
Yes, that is correct.  We have recently created a tutorial in the RapidWright documentation specifically around this task here: https://www.rapidwright.io/docs/Use_DREAMPlaceFPGA_via_FPGA_Interchange_Format.html

pwang7, 2023-10-17:
> Yes, that is correct. We have recently created a tutorial in the RapidWright documentation specifically around this task here: https://www.rapidwright.io/docs/Use_DREAMPlaceFPGA_via_FPGA_Interchange_Format.html

Thanks Chris!

After generating VU3P device FPGAIF file, when I run DREAMPlaceFPGA for `gnl_2_4_7_3.0_gnl_3500_03_7_80_80` following the tutorial, DREAMPlaceFPGA will end up with core dump.

But if I run DREAMPlaceFPGA with its build-in example `gnl_2_4_3_1.3_gnl_3000_07_3_80_80`, DREAMPlaceFPGA runs successfully.

BTW, have you heard of [OpenPARF](https://github.com/PKU-IDEA/OpenPARF)? it's a GPU-accelerated FPGA placer from the original author of DREAMPlace.

clavin-xlnx, 2023-10-17:
> After generating VU3P device FPGAIF file, when I run DREAMPlaceFPGA for `gnl_2_4_7_3.0_gnl_3500_03_7_80_80` following the tutorial, DREAMPlaceFPGA will end up with core dump.

Could you share the `.json` file you used and the log from the run?  
 
> BTW, have you heard of [OpenPARF](https://github.com/PKU-IDEA/OpenPARF)? it's a GPU-accelerated FPGA placer from the original author of DREAMPlace.

Yes, this is interesting work.

pwang7, 2023-10-17:
> > After generating VU3P device FPGAIF file, when I run DREAMPlaceFPGA for `gnl_2_4_7_3.0_gnl_3500_03_7_80_80` following the tutorial, DREAMPlaceFPGA will end up with core dump.
> 
> Could you share the `.json` file you used and the log from the run?

Below please find the json and the screen log files:
[gnl_2_4_7_3.0_gnl_3500_03_7_80_80.json](https://github.com/Xilinx/RapidWright/files/12936824/gnl_2_4_7_3.0_gnl_3500_03_7_80_80.json)
[gnl_2_4_7_3.0_gnl_3500_03_7_80_80.log](https://github.com/Xilinx/RapidWright/files/12936825/gnl_2_4_7_3.0_gnl_3500_03_7_80_80.log)

One caveat is I'm running DREAMPlaceFPGA inside a Docker on my desktop, and here is the [Dockerfile](https://github.com/pwang7/rw-jython/blob/master/scripts/DREAMPlaceFPGA.Dockerfile) I used to build and run DREAMPlaceFPGA.

clavin-xlnx, 2023-10-17:
> One caveat is I'm running DREAMPlaceFPGA inside a Docker on my desktop, and here is the [Dockerfile](https://github.com/pwang7/rw-jython/blob/master/scripts/DREAMPlaceFPGA.Dockerfile) I used to build and run DREAMPlaceFPGA.

Could you try using PyTorch 1.6/1.7/1.8?  We found some incompatibilities with PyTorch 2.0 which I know is different than 1.13, but perhaps something else might have changed.

clavin-xlnx, 2023-10-17:
@pwang7 - I've been able to reproduce the issue and I was able to solve the segfault by pulling in changes from the `lut0_support` branch in DREAMPlaceFPGA:

```
cd DREAMPlaceFPGA
git checkout lut0_support
git merge main
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$(dirname `pwd`)
make
make install
cd ..
# Try again
```

This resolved the segmentation fault for me.

pwang7, 2023-10-19:
> @pwang7 - I've been able to reproduce the issue and I was able to solve the segfault by pulling in changes from the `lut0_support` branch in DREAMPlaceFPGA:
>
> ```
> cd DREAMPlaceFPGA
> git checkout [...]

Thank you so much, @clavin-xlnx !
After switch to `lut0_support` branch, DREAMPlaceFPGA works very smooth on the tutorial example `gnl_2_4_7_3.0_gnl_3500_03_7_80_80` with PyTorch 1.13.1.

Just one minor issue, I cannot merge `lut0_support` with the `main` branch of DREAMPlaceFPGA, there are some [conflicts](https://github.com/rachelselinar/DREAMPlaceFPGA/compare/lut0_support).

pwang7, 2023-10-21:
I noticed that `lut0_support` is [merged](https://github.com/rachelselinar/DREAMPlaceFPGA/pull/24), now the main branch of DREAMPlaceFPGA can build and work successfully.

Thanks @clavin-xlnx !

eddieh-xlnx, 2023-10-25:
Looks like this issue is resolved now. Thanks @pwang7 for working with us on it!

closed 2023-10-25 by eddieh-xlnx