Xilinx/RapidWright #861: Generate dcp through custom define file

the-centry, 2023-10-25:
How to transfer the file to dcp file?
the target device is xcvu3p-ffvc1517-1-i
the nets' key is source tile.wire and sink tile.wire
[json file](https://drive.google.com/file/d/10EKYkS0M93x2TjSVRzgtxC2lIAoyFwqp/view?usp=sharing)

clavin-xlnx, 2023-10-27:
You'll want to use a few different APIs after parsing the JSON, for example (this code won't compile):

```java
Design design = new Design("designFromJSON", "xcvu3p-ffvc1517-1-i");
Device device = design.getDevice();

JSONObject obj = new JSONObject(new JSONTokener(new FileReader("result.log.json")));
JSONObject cellTiles = obj.getJSONObject("cells");
for (String cellTile : cellTiles.keySet()) {
    JSONObject jsonTile = cellTiles.getJSONObject(cellTile);
    Tile tile = device.getTile(cellTile);
    for (String belName : jsonTile.keySet()) {
        JSONObject attrs = jsonTile.getJSONObject(belName);
        Unisim cellType = Unisim.valueOf((String) attrs.get("primitive"));
        Site site = tile.getSites()[0]; // TODO This is a flaw in the schema, needs more information about which site
        BEL bel = null;
        
        // TODO - schema is mixing up sites and BELs
        if (belName.startsWith("IOB")) {
            site = device.getSite(belName);
        } else {
            bel = site.getBEL(belName);    
        }
        
        Cell c = null;
        if (bel == null) {
             // Its an IOB
            // TODO missing package pin name
            PinType pinType = cellType == Unisim.IBUF ? PinType.IN : PinType.OUT;
            c = design.createAndPlaceIOB("", pinType, pkgPin, "LVCMOS33");
        } else {
            c = design.createAndPlaceCell(design.getTopEDIFCell(), "cell_" + site + "_" + bel, cellType, site, bel);                    
        }
...
```

But it looks like there is some missing information in the JSON schema or it is inconsistent.  Is this JSON from a tool you are using or something you created?  I think it would have to be supplemented with more information to fully re-create a design, for example:
 * Site names
 * Consistency of Site vs BEL (`ALUT` is not a BEL, it should be `A6LUT` or `A5LUT`)
 * IOStandard used for IOBs
 * Package pin name
 * Logical to Physical Pin maps
 * ... (there might be a few more)

the-centry, 2023-10-28:
> ```java
> .
> ```

Year, the data is created according to my cognition of rapidwright! So the data may loss some information, if it is possible, you could change the json., such as ALUT->A6LUT, I0...I5-> A1...A6

the-centry, 2023-10-31:
> You'll want to use a few different APIs after parsing the JSON, for example (this code won't compile):
>
> ```java
> Design design = new Design("designFromJSON", "xcvu3p-ffvc1517-1-i");
> Device device = [...]

I have changed the json data and create cell successful. How to create the net with part pip defined by custom?  And I find some bugs:
1.pkgpin to site, why not site to pkgpin;
2.net could't add sink sitepin;
3. route net could't add pips as constrain;
Thanks so much!

the-centry, 2023-10-31:
`
public void DcpGenerate()
	{
		Design design = new Design("designFromJSON", "xcvu3p-ffvc1517-1-i");
		Device device = design.getDevice();
		for(BFDSCell item:m_BFDS.m_vCells)
		{
			String CellTile = item.m_sTileName;
			String CellSite = item.m_sSiteName;
			String CellName = item.m_sCellName;
			String CellPrimitive = item.m_sPrimitive;
			Unisim cellType = Unisim.valueOf(CellPrimitive);
			Tile tile = device.getTile(CellTile);
			Site site = tile.getSites()[0];
			BEL bel = null;
			if(CellSite!=null && CellSite.startsWith("IOB"))
			{
				site = device.getSite(CellSite);
			}
			else {
				bel = site.getBEL(item.m_sBelName);
			}
			Cell c = null;
			if(bel == null)
			{
				PinType pinType = cellType == Unisim.IBUF ? PinType.IN : PinType.OUT;
				String pkgpin = item.m_sPkgpin;
				 c = design.createAndPlaceIOB(CellName, pinType, pkgpin, "LVCMOS18");
			}else {
	            c = design.createAndPlaceCell(design.getTopEDIFCell(), CellName, cellType, site, bel);                    
	        }
			System.out.println(item.m_sCellName);
		}
		for(BFDSNet item:m_BFDS.m_vNets)
		{
			String netname  =item.m_sNetName;
			String srctile = item.m_sSourceTile;
			String sourcewire = item.m_sSourceWire;
			String dsttile = item.m_sSinkTile;
			String sinkwire = item.m_sSinkWire;
			Wire srcWire = new Wire(device, srctile, sourcewire);
			//RouteNode src = new RouteNode(device.getTile(srctile), srcWire.getWireIndex());
			Wire dstWire = new Wire(device, dsttile, sinkwire);
			//RouteNode sink = new RouteNode(device.getTile(dsttile), dstWire.getWireIndex());
			Net net = design.createNet(netname);
			SitePin srcPin = srcWire.getSitePin();
			SiteInst srcSiteInst = design.getSiteInstFromSite(srcPin.getSite());
			net.setSource(srcSiteInst.getSitePinInst(srcPin.getPinName()));
			SitePin dstPin = dstWire.getSitePin();
			SiteInst dstSiteInst = design.getSiteInstFromSite(dstPin.getSite());
			net.setSource(dstSiteInst.getSitePinInst(dstPin.getPinName()));
			for(String pipname:item.m_vPips)
			{
				String[] pipInfo = pipname.split("\\.");
				String pipTileName = pipInfo[0];
				String pipDst = pipInfo[1];
				String pipSrc = pipInfo[2];
				Tile pipTile = device.getTile(pipTileName);
				net.addPIP(pipTile.getPIP(pipTile.getWireIndex(pipSrc), pipTile.getWireIndex(pipDst)));
			}	
		}
		String dcppath = "D:\\softcache\\result.log.dcp";
		design.writeCheckpoint(dcppath);	
	}
`

the-centry, 2023-10-31:
``` 
public void DcpGenerate()
	{
		Design design = new Design("designFromJSON", "xcvu3p-ffvc1517-1-i");
		Device device = design.getDevice();
		for(BFDSCell item:m_BFDS.m_vCells)
		{
			String CellTile = item.m_sTileName;
			String CellSite = item.m_sSiteName;
			String CellName = item.m_sCellName;
			String CellPrimitive = item.m_sPrimitive;
			Unisim cellType = Unisim.valueOf(CellPrimitive);
			Tile tile = device.getTile(CellTile);
			Site site = tile.getSites()[0];
			BEL bel = null;
			if(CellSite!=null && CellSite.startsWith("IOB"))
			{
				site = device.getSite(CellSite);
			}
			else {
				bel = site.getBEL(item.m_sBelName);
			}
			Cell c = null;
			if(bel == null)
			{
				PinType pinType = cellType == Unisim.IBUF ? PinType.IN : PinType.OUT;
				String pkgpin = item.m_sPkgpin;
				 c = design.createAndPlaceIOB(CellName, pinType, pkgpin, "LVCMOS18");
			}else {
	            c = design.createAndPlaceCell(design.getTopEDIFCell(), CellName, cellType, site, bel);                    
	        }
			System.out.println(item.m_sCellName);
		}
		for(BFDSNet item:m_BFDS.m_vNets)
		{
			String netname  =item.m_sNetName;
			String srctile = item.m_sSourceTile;
			String sourcewire = item.m_sSourceWire;
			String dsttile = item.m_sSinkTile;
			String sinkwire = item.m_sSinkWire;
			Wire srcWire = new Wire(device, srctile, sourcewire);
			//RouteNode src = new RouteNode(device.getTile(srctile), srcWire.getWireIndex());
			Wire dstWire = new Wire(device, dsttile, sinkwire);
			//RouteNode sink = new RouteNode(device.getTile(dsttile), dstWire.getWireIndex());
			Net net = design.createNet(netname);
			SitePin srcPin = srcWire.getSitePin();
			SiteInst srcSiteInst = design.getSiteInstFromSite(srcPin.getSite());
			net.setSource(srcSiteInst.getSitePinInst(srcPin.getPinName()));
			SitePin dstPin = dstWire.getSitePin();
			SiteInst dstSiteInst = design.getSiteInstFromSite(dstPin.getSite());
			net.setSource(dstSiteInst.getSitePinInst(dstPin.getPinName()));
			for(String pipname:item.m_vPips)
			{
				String[] pipInfo = pipname.split("\\.");
				String pipTileName = pipInfo[0];
				String pipDst = pipInfo[1];
				String pipSrc = pipInfo[2];
				Tile pipTile = device.getTile(pipTileName);
				net.addPIP(pipTile.getPIP(pipTile.getWireIndex(pipSrc), pipTile.getWireIndex(pipDst)));
			}	
		}
		String dcppath = "D:\\softcache\\result.log.dcp";
		design.writeCheckpoint(dcppath);	
	}
```

clavin-xlnx, 2023-10-31:
@the-centry Good progress!  

Regarding

> 1.pkgpin to site, why not site to pkgpin

That is a good point, we don't have an easy way to go from site to package pin name, here is some code that can build a map for you to do a lookup by site to get the package pin name:

```java
Map<Site, String> siteToPackagePinName = new HashMap<>();
for (Entry<String, PackagePin> p : device.getActivePackage().getPackagePinMap().entrySet()) {
    Site site = p.getValue().getSite();
    if (site != null) {
        siteToPackagePinName.put(site, p.getKey());
    }
}
// ... lookup package pin name by site
// siteToPackagePinName.get(site)
```

> 2.net could't add sink sitepin;

You actually will need to create objects called `SitePinInst`, here is an example:

```java
SitePin srcPin = srcWire.getSitePin();
SiteInst srcSiteInst = design.getSiteInstFromSite(srcPin.getSite());
net.createPin(srcPin.getName(), srcSiteInst);
```
see javadocs for details: https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Net.html#createPin-java.lang.String-com.xilinx.rapidwright.design.SiteInst-

> 3. route net could't add pips as constrain;

The APIs you are using are correct, but the input data is not consistent to create the `PIP` objects.  Each `PIP` describes the connection between two wires in the same `Tile`.  However, often in Vivado we describe routing with `Node` objects which are not contained to a tile (see this description of `Node` in our documentation https://www.rapidwright.io/docs/Xilinx_Architecture.html#node).  Perhaps if you could share a specific instance of where it isn't working we can look at it in detail.

the-centry, 2023-10-31:
@clavin-xlnx Thanks so much!
`1.However, often in Vivado we describe routing with Node objects which are not contained to a tile` 
It means that I transfer pip's wire to node, route could use these node as constrain?
Second:
The problem I found that the cells are empty in dcp file while use the way that you give!
The generated dcp as [result.dcp](https://drive.google.com/file/d/11ps_c5Er6uZ6UIvFqeOwftrbus8Ne0sX/view?usp=sharing)
`The code is`:
```
package SB2RP;
import com.xilinx.rapidwright.design.ConstraintGroup;
import com.xilinx.rapidwright.design.NetType;
import com.xilinx.rapidwright.design.SiteInst;
import com.xilinx.rapidwright.device.Site;
import com.xilinx.rapidwright.device.BEL;
import com.xilinx.rapidwright.device.Wire;
import com.xilinx.rapidwright.device.PIP;
import com.xilinx.rapidwright.edif.EDIFCell;
import com.xilinx.rapidwright.edif.EDIFCellInst;
import com.xilinx.rapidwright.edif.EDIFNet;
import com.xilinx.rapidwright.edif.EDIFPortInst;
import com.xilinx.rapidwright.edif.EDIFTools;
import com.xilinx.rapidwright.rwroute.RWRoute;
import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.SitePinInst;
import com.xilinx.rapidwright.design.PinType;
import com.xilinx.rapidwright.design.Unisim;
import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.SitePin;
import com.xilinx.rapidwright.device.TileTypeEnum;
import com.xilinx.rapidwright.device.Tile;
import com.xilinx.rapidwright.router.Router;
import com.xilinx.rapidwright.router.RouteNode;

public class BFDSDCP {
	public BFDSParser m_BFDS;
	public void InitBFDS(String json)
	{
		m_BFDS = new BFDSParser();
		m_BFDS.FileParser(json);
	}
	public void DcpGenerate()
	{
		Design design = new Design("designFromJSON", "xcvu3p-ffvc1517-1-i");
		Device device = design.getDevice();
		for(BFDSCell item:m_BFDS.m_vCells)
		{
			String CellTile = item.m_sTileName;
			String CellSite = item.m_sSiteName;
			String CellName = item.m_sCellName;
			String CellPrimitive = item.m_sPrimitive;
			Unisim cellType = Unisim.valueOf(CellPrimitive);
			Tile tile = device.getTile(CellTile);
			Site site = tile.getSites()[0];
			BEL bel = null;
			if(CellSite!=null && CellSite.startsWith("IOB"))
			{
				site = device.getSite(CellSite);
			}
			else {
				bel = site.getBEL(item.m_sBelName);
			}
			Cell c = null;
			if(bel == null)
			{
				PinType pinType = cellType == Unisim.IBUF ? PinType.IN : PinType.OUT;
				String pkgpin = item.m_sPkgpin;
				 c = design.createAndPlaceIOB(CellName, pinType, pkgpin, "LVCMOS18");
			}else {
	            c = design.createAndPlaceCell(design.getTopEDIFCell(), CellName, cellType, site, bel);                    
	        }
			System.out.println(item.m_sCellName);
		}
		for(BFDSNet item:m_BFDS.m_vNets)
		{
			String netname  =item.m_sNetName;
			String srctile = item.m_sSourceTile;
			String sourcewire = item.m_sSourceWire;
			String dsttile = item.m_sSinkTile;
			String sinkwire = item.m_sSinkWire;
			Wire srcWire = new Wire(device, srctile, sourcewire);
			//RouteNode src = new RouteNode(device.getTile(srctile), srcWire.getWireIndex());
			Wire dstWire = new Wire(device, dsttile, sinkwire);
			//RouteNode sink = new RouteNode(device.getTile(dsttile), dstWire.getWireIndex());
			Net net = design.createNet(netname);
			SitePin srcPin = srcWire.getSitePin();
			SiteInst srcSiteInst = design.getSiteInstFromSite(srcPin.getSite());
			SitePinInst srcSitePinInst =  new SitePinInst(srcPin.getPinName(),srcSiteInst);//srcSiteInst.getSitePinInst(srcPin.getPinName());
			net.setSource(srcSitePinInst);
			SitePin dstPin = dstWire.getSitePin();
			SiteInst dstSiteInst = design.getSiteInstFromSite(dstPin.getSite());
			SitePinInst dstSitePinInst = new SitePinInst(dstPin.getPinName(),dstSiteInst);
			net.addPin(dstSitePinInst);
			//net.setSource(dstSiteInst.getSitePinInst(dstPin.getPinName()));
			for(String pipname:item.m_vPips)
			{
				String[] pipInfo = pipname.split("\\.");
				String pipTileName = pipInfo[0];
				String pipDst = pipInfo[1];
				String pipSrc = pipInfo[2];
				Tile pipTile = device.getTile(pipTileName);
				net.addPIP(pipTile.getPIP(pipTile.getWireIndex(pipSrc), pipTile.getWireIndex(pipDst)));
			}	
		}
		String dcppath = "D:\\softcache\\result0.log.dcp";
		design.routeSites();
//		design.updateDesignWithCheckpointPlaceAndRoute(dcppath);
		design.writeCheckpoint(dcppath);
		
	}
	public static void main(String[] args) {
		// TODO Auto-generated method stub
		String file="D:\\softcache\\result.log.json";
		BFDSDCP test = new BFDSDCP();
		test.InitBFDS(file);
		test.DcpGenerate();
	}

}
```

clavin-xlnx, 2023-10-31:
@the-centry - if you would like me to try and diagnose the issue by running your code, I am missing some of the dependencies (`BFDSParser`, `BFDSCell`, `BFDSNet`, for example).  Also, if you could provide the revised version of your json file, that would also be needed.

the-centry, 2023-11-01:
@clavin-xlnx Ok, Thanks!
The code and the file is here [test data and code](https://drive.google.com/file/d/1ZwxesCgQiZRhVaL_dp1Q8ZChvrViCmDb/view?usp=drive_link). My first tring of java!

closed 2023-11-03 by the-centry