Xilinx/RapidWright #873: How to use outmuxa?

the-centry, 2023-11-07:
How to use the outmux(a|b|c|d) while create a cell?
Thanks so much!

the-centry, 2023-11-07:
WARNING: Could not find parent of net "CLEL_R_X73Y0.CLE_CLE_L_SITE_0_A_O.HPIO_L_X71Y180.HPIO_IOBPAIR_72_OP_PIN", please check that the netlist is fully connected through all levels of hierarchy for this net.

clavin-xlnx, 2023-11-08:
> How to use the outmux(a|b|c|d) while create a cell? Thanks so much!

```java
package com.xilinx.rapidwright.examples;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.SitePinInst;
import com.xilinx.rapidwright.design.Unisim;
import com.xilinx.rapidwright.device.BELPin;
import com.xilinx.rapidwright.rwroute.RWRoute;

public class LUT1Example {

    public static void main(String[] args) {
        Design design = new Design("example", "xcvu3p-ffvc1517-1-i");
        Cell cell = design.createAndPlaceCell("lut1", Unisim.LUT1, "SLICE_X18Y31/A6LUT");
        cell.addProperty("INIT", "O=I0");
        Net net = design.createNet("net0");
        net.connect(cell, "O");
        net.connect(cell, "I0");
        // By default, it will use the A_O output pin, so we'll remove it and use AMUX
        // instead to force the use of OUTMUXA
        net.removeSource();
        SitePinInst outputPin = net.createPin("AMUX", cell.getSiteInst());
        BELPin lutOutput = cell.getBEL().getPin("O6");
        // Route the intra-site connection
        cell.getSiteInst().routeIntraSiteNet(net, lutOutput, outputPin.getBELPin());
        // Route the inter-site connection
        RWRoute.routeDesignFullNonTimingDriven(design);
        design.writeCheckpoint("example.dcp");
    }
}
```
This will force the use of the OUTMUX as shown in this routed view in Vivado:
[image]

clavin-xlnx, 2023-11-08:
> WARNING: Could not find parent of net "CLEL_R_X73Y0.CLE_CLE_L_SITE_0_A_O.HPIO_L_X71Y180.HPIO_IOBPAIR_72_OP_PIN", please check that the netlist is fully connected through all levels of hierarchy for this net.

This means there is an inconsistency in the created logical netlist.  You may have a logical net name that is inconsistent with the physical net name.

the-centry, 2023-11-08:
> This means there is an inconsistency in the created logical netlist. You may have a logical net name that is inconsistent with the physical net name.

Whether it means I need connect the cell and sitepin while create a cell instance!

closed 2023-11-09 by the-centry