Xilinx/RapidWright #956: pip.getEndNode returning null

trharoldsen, 2024-02-09:
I'm get a null return value from `PIP.getEndNode()` for a PIP obtained from `Node.getAllDownhillPIPs()`.  The original node is `INT_R_X3Y394/INT_R.NN6END3` and the failing PIP is `INT_R_X3Y394/INT_R.NN6END3->>LVB0`.  The part in question is xc7vx1140tflg1926-1.

I'm not positive, but I believe the issue is coming in the Node.isInvalidNode() call inside the Node.getNode() function.

The issue is observed in both versions 2022.1.1 and 2023.2.1.

**Code to reproduce:**

```java
import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.PIP;
import com.xilinx.rapidwright.device.Node;

public class Test {

    public static void main(String[] args) {
        Device device = Device.getDevice("xc7vx1140tflg1926-1");

        Node n = device.getNode("INT_R_X3Y394/NN6END3");
        for (PIP p : n.getAllDownhillPIPs()) {
            System.out.println(p + ", " + p.getEndNode());
        }

        System.out.println();

        PIP p = device.getPIP("INT_R_X3Y394/INT_R.NN6END3->>LVB0");
        System.out.println(p);
        System.out.println(p.getEndWire());
        System.out.println(p.getEndWireName());
        System.out.println(p.getEndWireIndex());
        System.out.println(p.getStartNode());
        System.out.println(p.getEndNode());
    }
}
```

**Observed Output:**

```
INT_R_X3Y394/INT_R.NN6END3->>WW4BEG3, INT_R_X3Y394/WW4BEG3
INT_R_X3Y394/INT_R.NN6END3->>SE6BEG3, INT_R_X3Y394/SE6BEG3
INT_R_X3Y394/INT_R.NN6END3->>NW6BEG3, INT_R_X3Y394/NW6BEG3
INT_R_X3Y394/INT_R.NN6END3->>NW2BEG3, INT_R_X3Y394/NW2BEG3
INT_R_X3Y394/INT_R.NN6END3->>NR1BEG3, INT_R_X3Y394/NR1BEG3
INT_R_X3Y394/INT_R.NN6END3->>NN6BEG3, INT_R_X3Y394/NN6BEG3
INT_R_X3Y394/INT_R.NN6END3->>NN2BEG3, INT_R_X3Y394/NN2BEG3
INT_R_X3Y394/INT_R.NN6END3->>NE6BEG3, INT_R_X3Y394/NE6BEG3
INT_R_X3Y394/INT_R.NN6END3->>NE2BEG3, INT_R_X3Y394/NE2BEG3
INT_R_X3Y394/INT_R.NN6END3->>EE4BEG3, INT_R_X3Y394/EE4BEG3
INT_R_X3Y394/INT_R.NN6END3->>EE2BEG3, INT_R_X3Y394/EE2BEG3
INT_R_X3Y394/INT_R.NN6END3->>WW2BEG2, INT_R_X3Y394/WW2BEG2
INT_R_X3Y394/INT_R.NN6END3->>SR1BEG3, INT_R_X3Y394/SR1BEG3
INT_R_X3Y394/INT_R.NN6END3->>NL1BEG2, INT_R_X3Y394/NL1BEG2
INT_R_X3Y394/INT_R.NN6END3->>EL1BEG2, INT_R_X3Y394/EL1BEG2
INT_R_X3Y394/INT_R.NN6END3->>WR1BEG_S0, INT_R_X3Y394/WR1BEG_S0
INT_R_X3Y394/INT_R.NN6END3->>LVB12, INT_R_X3Y394/LVB12
INT_R_X3Y394/INT_R.NN6END3->>LVB0, null

INT_R_X3Y394/INT_R.NN6END3->>LVB0
INT_R_X3Y394/LVB0
LVB0
161
INT_R_X3Y388/NN6BEG3
null
```

**Expected outcome:**

PIP.getEndNode() for all PIPs retrieved from Node.getAllDownhillPIPs() should return a non-null node.

clavin-xlnx, 2024-02-10:
Thanks for the complete test case.  However, this behavior is consistent with Vivado.  The wire `INT_R_X3Y394/LVB0` is a wire object but it doesn't belong to a `Node` object.  This is a rare case that occurs in some devices/architectures.  If you run equivalent commands in Vivado Tcl:
```
# Getting the wire, confirming it is present
$ get_wires INT_R_X3Y394/LVB0
INT_R_X3Y394/LVB0
```
```
# Attempting to get the non-existent node
$ get_nodes INT_R_X3Y394/LVB0
WARNING: [Vivado 12-2683] No nodes matched 'get_nodes INT_R_X3Y394/LVB0'
```
Another way to confirm this is by querying the PIP for its nodes.  
```
# For a regular node, we get two results
get_nodes -of [get_pips INT_R_X3Y394/INT_R.NN6END3->>LVB12]
INT_R_X3Y394/LVB12 INT_R_X3Y388/NN6BEG3
```
However, for the PIP in question, we only get one result:
```
get_nodes -of [get_pips INT_R_X3Y394/INT_R.NN6END3->>LVB0]
INT_R_X3Y388/NN6BEG3
```

Hope that helps.

trharoldsen, 2024-02-10:
Thanks for the thorough explanation, Chris.

closed 2024-02-11 by trharoldsen