Xilinx/RapidWright #957: Odd behavior with Wire constructor and Node.isInvalidNode

trharoldsen, 2024-02-16:
Relating to issue #956, I observed this odd behavior with the interchange between the Wire constructor and Node.isInvalidNode.  I have a sample code below showing the issue.

In summary, when creating a wire from an invalid node and then reconverting the created wire to a node, the reformed node is no longer indicated as an invalid node even though by all accounts it feels like it should be.  

I believe the issue is stemming from a bit length truncation that happens inside the Wire constructor.  An invalid node has a wire value of -1 (32 bits).  Inside the Wire constructor, the wire value is truncated to 16 bits with a masking operation.  After the truncation, an invalid wire has a value of 65535 (-1 in 16 bits).  The Node.isInvalidNode function looks at both the tile and the wire index and reports the Node as invalid only if the tile is null and the wire index is -1 (32 bits).

Solution-wise, it feels that (1) the wire constructor should check for a -1 value and bypass the truncation if so and (2) Node.isInvalidNode should return true if either the tile is null or the wire is -1 (though this admittedly is a much more significant change to the Node API).

```
package edu;

import com.xilinx.rapidwright.device.*;

public class Test {

    public static void main(String[] args) {
        Device device = Device.getDevice("xc7vx1140tflg1926-1");

        Node n = device.getNode("INT_R_X3Y394/NN6END3");
        Node d1 = null;
        for (Node d : n.getAllDownhillNodes()) {
            if (d.isInvalidNode())
                d1 = d;
        }
        
        Wire w = new Wire(n.getTile(), n.getWire());
        Node d2 = w.getNode();
        System.out.println(d1.equals(d2));
        System.out.println(d1.isInvalidNode());
        System.out.println(d2.isInvalidNode());
    }
}
```

Output
```
false
true
false
```

Expected:
d2.isInvalidNode() would return true.

clavin-xlnx, 2024-02-16:
Thanks for the code example, however, if I run the code above verbatim, I get a `NullPointerException` on the line `System.out.println(d1.equals(d2));` because `d1` is null because no nodes from `n.getAllDownhillNodes()` are invalid.  One interesting note is that `INT_R_X3Y394/NN6END3` is not a canonical node, its a wire that belongs to the node called `INT_R_X3Y388/NN6BEG3`.

Could you perhaps check the code again in the case there is something missing?

trharoldsen, 2024-02-16:
I apologize.  I'm switching between 2022.1.1 and 2023.2.1 because of some incompatible API differences between the two.  I apparently missed updating the jar on the previous example when validating on the latest version.

It seems this issue has been fixed with 2023.2.1 as Node.getAllDownhillNodes() now filters these invalid nodes and PIP.getEndNode() returns null instead of an invalid node for these PIPs.

Closing as fixed.

closed 2024-02-16 by trharoldsen