Xilinx/RapidWright #959: It's better for `netlist.getTopCellInst().getPortInsts()` to not return empty list.

zhuguiyuan, 2024-02-18:
```java
var netlist = EDIFTools.loadEDIFFile(netlistFile);

// The nets contain some portInsts of topCellInst.
netlist.getTopCell().getNets().forEach(n -> System.out
   .println("name: " + n.getName() + " port_insts: " + n.getPortInsts()));

// However, we cannot directly get the portInsts of the top cellInst.
// This method return an empty list.
netlist.getTopCellInst().getPortInsts().forEach(p -> System.out.println(p.getName()));
```

Since the top cell inst is an `EDIFCellInst`, it's a better idea to attach its `EDIFPortInst` onto it.

zhuguiyuan, 2024-02-18:
Here is an example:

EDIF File
```edif
(edif one_one_detector
  (edifversion 2 0 0)
  (edifLevel 0)
  (keywordmap (keywordlevel 0))
(status
 (written
  (timeStamp 2024 02 13 14 25 56)
  (program "Vivado" (version "2023.2.1"))
  (comment "Built on 'Thu Dec 14 12:24:51 MST 2023'")
  (comment "Built by 'xbuild'")
 )
)
  (Library hdi_primitives
    (edifLevel 0)
    (technology (numberDefinition ))
   (cell BUFG (celltype GENERIC)
     (view netlist (viewtype NETLIST)
       (interface 
        (port O (direction OUTPUT))
        (port I (direction INPUT))
       )
     )
   )
   (cell IBUF (celltype GENERIC)
     (view netlist (viewtype NETLIST)
       (interface 
        (port O (direction OUTPUT))
        (port I (direction INPUT))
       )
     )
   )
   (cell GND (celltype GENERIC)
     (view netlist (viewtype NETLIST)
       (interface 
        (port G (direction OUTPUT))
       )
     )
   )
   (cell VCC (celltype GENERIC)
     (view netlist (viewtype NETLIST)
       (interface 
        (port P (direction OUTPUT))
       )
     )
   )
   (cell LUT3 (celltype GENERIC)
     (view netlist (viewtype NETLIST)
       (interface 
        (port O (direction OUTPUT))
        (port I0 (direction INPUT))
        (port I1 (direction INPUT))
        (port I2 (direction INPUT))
       )
     )
   )
   (cell FDRE (celltype GENERIC)
     (view netlist (viewtype NETLIST)
       (interface 
        (port Q (direction OUTPUT))
        (port C (direction INPUT))
        (port CE (direction INPUT))
        (port D (direction INPUT))
        (port R (direction INPUT))
       )
     )
   )
[... 308 lines cut ...]
Java File
```java.
    var netlist = EDIFTools.loadEDIFFile(netlistFile);
    netlist.getTopCell().getNets().forEach(n -> System.out
        .println("name: " + n.getName() + " port_insts: " + n.getPortInsts()));
    System.out.println(netlist.getTopCellInst().getPortInsts().size());
```

Output:
```
name: clk port_insts: [clk, clk_IBUF_inst/I]
name: in_IBUF[0] port_insts: [in_IBUF[0]_inst/O, inst0/D[0]]
name: clk_IBUF_BUFG port_insts: [clk_IBUF_BUFG_inst/O, inst0/CLK, inst1/CLK]
name: out[1] port_insts: [out[1], out_OBUF[1]_inst/O]
name: out[0] port_insts: [out[0], out_OBUF[0]_inst/O]
name: clk_IBUF port_insts: [clk_IBUF_BUFG_inst/I, clk_IBUF_inst/O]
name: in_IBUF[1] port_insts: [in_IBUF[1]_inst/O, inst1/D[0]]
name: rst port_insts: [rst, rst_IBUF_inst/I]
name: out_OBUF[0] port_insts: [inst0/out_OBUF[0], out_OBUF[0]_inst/I]
name: out_OBUF[1] port_insts: [inst1/out_OBUF[0], out_OBUF[1]_inst/I]
name: rst_IBUF port_insts: [inst0/SR[0], inst1/SR[0], rst_IBUF_inst/O]
name: in[0] port_insts: [in[0], in_IBUF[0]_inst/I]
name: in[1] port_insts: [in[1], in_IBUF[1]_inst/I]
0
```

clavin-xlnx, 2024-02-20:
Thank you for the post and example.  However, I am not sure if I agree with the need to attach `EDIFPortInst` objects to the top `EDIFCellInst`.  The top instance is a special case and is only present to instantiate the design.  It has no outer context and doesn't connect to anything larger than itself.  The top cell can have internal facing `EDIFPortInst` objects connecting to its internal nets, but that would not be present on the top instance.  

Can you share more about your use case on what you are trying to accomplish?  Perhaps we can help in another way.

eddieh-xlnx, 2025-05-13:
Closing due to inactivity. Please re-open or file a new issue if there are additional concerns.