Xilinx/RapidWright #963: Vivado not preserving the placement of DSP48E2 and RAMB18E2 cells

oshears, 2024-02-27:
I'm experiencing an issue in RapidWright 2023.2.1 where the placement of some cells in a design are not preserved when I open a checkpoint, exported by RapidWright, into Vivado 2023.2. The cells that are getting unplaced are `DSP48E2` and `RAMB18E2` cells on a Zynq UltraScale+ device (xczu3eg-sfva625-1-e).

In this scenario, I have an initial design checkpoint that I created with Vivado. I can see that all cells are placed in this checkpoint via Vivado (shown below):

[image]

With this checkpoint (`checkpoint-1.dcp`), I run the following RapidWright code, which does not modify the design:

```java
import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;

public class Demo {
    public static void main(String[] args){
        System.out.println("Reading checkpoint.");
        Design design = Design.readCheckpoint("checkpoint-1.dcp");

        Cell bramCell = design.getCell("harness/state_reg");
        System.out.println(bramCell);
        System.out.println(bramCell.isPlaced());
        System.out.println(bramCell.getSite());

        Cell dspCell = design.getCell("harness/counter_reg/DSP_ALU_INST");
        System.out.println(dspCell);
        System.out.println(dspCell.isPlaced());
        System.out.println(dspCell.getSite());

        System.out.println("Writing checkpoint.");
        design.writeCheckpoint("checkpoint-2.dcp");
    }
}
```

The RapidWright code indicates that the placement is still present for the DSP and BRAM cells. After exporting the checkpoint (`checkpoint-2.dcp`) with RapidWright, opening the new checkpoint in Vivado results in the `DSP48E2` and `RAMB18E2` cells being unplaced. This is shown in the screenshot from Vivado below:

[image]

Do you all have any idea why this behavior might be occurring?

clavin-xlnx, 2024-02-27:
Could you provide an input DCP that exhibits this behavior?

oshears, 2024-02-27:
Here's the sample DCP:

[dcp.tar.gz](https://github.com/Xilinx/RapidWright/files/14413614/dcp.tar.gz)

eddieh-xlnx, 2024-02-27:
It looks like these DSP/BRAM cells are not connected to anything, which is not the typical way of using cells. We've come across this before in https://github.com/Xilinx/RapidWright/discussions/634 -- the workaround there was to connect the cells to something so that they do propagate into Vivado correctly. Is this a possibility here, if this is just a intermediate point on the way to actually using those cells to do some computation?

eddieh-xlnx, 2024-02-27:
Oh wait. I can see from your screenshot that the cells are connected to something, and it's just the RapidWright code doesn't connect anything new. Carry on and ignore me!

clavin-xlnx, 2024-02-27:
Still working on it, but the short answer is, there are hierarchy separators in the name.  RapidWright is designed to handle these, but somewhere along the chain of custody, they have become confused.

A work around is to simply don't use `/` in the names of your cells/instances, it is bad practice and leads to confusion.  

I'll continue looking at why this is not being handled properly.

clavin-xlnx, 2024-02-27:
If I rename `harness/counter_reg` to `harness_counter_reg`, Vivado has no issues with the DSP placement.

closed 2024-02-27 by oshears

clavin-xlnx, 2024-02-27:
Just as a final word, the DCP format that RapidWright writes out is an older version that has a limitation where Vivado cannot restore placement of cells that contain hierarchy separators in their names.  We have stayed with the older version as it allows the widest compatibility of use amongst different Vivado versions.  We may offer a newer version of the DCP format in the future, but for now, this is simply a limitation and recommend that cell/instance names should not contain hierarchy separators.