Xilinx/RapidWright #964: `design.getBELAttrs()` returns None

dg-deus, 2024-02-29:
To work with RapidWright design pacakges in python, I implemented a simple mux, and placed it in a single site.
Then I created dcp file of it with the corresponding edif file as you can find in the attachments.
[mux16.zip](https://github.com/Xilinx/RapidWright/files/14443945/mux16.zip)

Then, I loaded it to test design pacakges. At this time, I wanted to query properties of used BELs (not Cells) in the site.
```
import jpype.imports
jpype.startJVM(classpath=["rapidwright-2023.2.1-standalone-win64.jar"])
RwDesign = Design.readCheckpoint('mux16.dcp', 'mux16.edif')
for RwSite, RwSiteConfig in RwDesign.getBELAttrs().items():
    print(f"{RwSite}, {RwSiteConfig}")
```

However, `RwDesign.getBELAttrs()` returns `None`.
How can I access BEL properties(attributes) used in the netlist of given checkpoint file?

Thank you in advance.

clavin-xlnx, 2024-02-29:
The `SiteConfig` and `BELAttr` objects are to support representation in the Versal device series,. 
 The DCP that you have provided targets a Series 7 device, so these constructs are not used (see discussion here: https://github.com/Xilinx/RapidWright/discussions/598). 

Can you describe some of the specific attributes you are looking for?  Most configuration exists on the `Cell` and/or `SiteInst` objects, so those are the likely candidates to query.

dg-deus, 2024-03-01:
I want to query configuration options of `BEL`s used to implement the given properties of `Cell`s in netlist.

clavin-xlnx, 2024-03-01:
> I want to query configuration options of `BEL`s used to implement the given properties of `Cell`s in netlist.

Unfortunately, there is not a comprehensive list of these properties, they mostly need to be queried or computed adhoc.

closed 2024-03-02 by dg-deus