Xilinx/RapidWright #967: make blackblox issue

pwang7, 2024-03-06:
I'm using [MakeBlackbox](https://github.com/Xilinx/RapidWright/blob/60a571446d1bb78b9aba6ed497914631c2e7bc97/src/com/xilinx/rapidwright/util/MakeBlackBox.java#L35) in my design following the tutorial [Reuse Timing-closed Logic As A Shell
](https://www.rapidwright.io/docs/ReusingTimingClosedLogicAsAShell.html).

In my design, I use two hard IPs: XDMA and CMAC. However, after I cut several blackboxes from my logic, some modules of XDMA and CMAC become blackbox too, such as `udma_wrapper`, `ram_top`, `i_cmac_usplus_0_top`, as depicted in the attached pictures:

[image]
[image]
[image]

clavin-xlnx, 2024-03-07:
RapidWright is unable to decrypt netlists and most IPs have their netlist encrypted.  Thus, when you load a DCP back into Vivado from RapidWright, you often cannot simply reload the DCP from RapidWright, but you need to use the Tcl script that gets generated when encrypted IPs are detected.  This way, the Tcl script is able to reassemble the encrypted portions back into the top level design.  If this is not done, the IPs will look like black boxes.  

When you write out the DCP from RapidWright, do you see a message about using a Tcl script to load the design back into Vivado?  The message may looks something like "...may contain encrypted cells. To correctly load the design into Vivado please source this Tcl script to open the checkpoint:..."

pwang7, 2024-03-07:
> RapidWright is unable to decrypt netlists and most IPs have their netlist encrypted. Thus, when you load a DCP back into Vivado from RapidWright, you often cannot simply reload the DCP from [...]

Thanks Chris! Now I can successfully load the shell DCP.

After load the shell DCP, I read back the DCP for the blackbox,
```Shell
read_checkpoint -cell <CELL_NAME> <BLACKBOX_SYNTH.dcp>
```
and tried to place design. However, Vivado gives me another error:
```Shell
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
```

I guess Vivado is in non-project mode after read checkpoint, then how to open design?

clavin-xlnx, 2024-03-08:
When you load the DCP from a Tcl script, Vivado will be in project mode.  You can try writing out a new DCP after loading it via Tcl and then re-opening the newly written DCP in Vivado and try that to see if it alleviates the error message you are seeing.   

However, this error is puzzling and makes me think that either the wrong window was active when you ran the `read_checkpoint -cell` command as you should not have to write out the DCP a second time in Vivado to make this work.

pwang7, 2024-03-08:
> When you load the DCP from a Tcl script, Vivado will be in project mode. You can try writing out a new DCP after loading it via Tcl and then re-opening the newly written DCP in Vivado and try that [...]

Thanks Chris! Writing out a new DCP after loading encrypted netlists via Tcl, it works.

However, when I use the latest version, `rapidwright-2023.2.1-standalone-lin64.jar`, `MakeBlackbox` reports the following runtime exception:
```
==============================================================================
==                               MakeBlackbox                               ==
==============================================================================
                Read DCP:     9.741s
Exception in thread "main" java.lang.RuntimeException: ERROR: Cell corresponding to pin 'cmac_inst/inst/i_cmac_usplus_0_top/ctl_rx_check_etype_gcp' not found.
        at com.xilinx.rapidwright.eco.ECOTools.connectNet(ECOTools.java:402)
        at com.xilinx.rapidwright.eco.ECOTools.createAndPlaceInlineCellOnInputPin(ECOTools.java:1171)
        at com.xilinx.rapidwright.design.DesignTools.prepareShellBlackBoxForRouting(DesignTools.java:4013)
        at com.xilinx.rapidwright.util.MakeBlackBox.main(MakeBlackBox.java:53)
```

Previously, I used `rapidwright-2023.2.0-standalone-lin64.jar`, and `MakeBlackbox` works.

clavin-xlnx, 2024-03-08:
> However, when I use the latest version, `rapidwright-2023.2.1-standalone-lin64.jar`, `MakeBlackbox` reports the following runtime exception:
>
> ``` [...]

Would you be able to provide an example design that triggers this error?  It appears that it might be a bug and a test case would make it much easier to reproduce and diagnose.

pwang7, 2024-03-11:
> Would you be able to provide an example design that triggers this error? It appears that it might be a bug and a test case would make it much easier to reproduce and diagnose.

I uploaded my design [post_route.dcp](https://drive.google.com/file/d/1a9y3jPBETBH9Zne0e0t-LUDemuFHmLLP/view?usp=sharing) to Google drive. The command to reproduce this exception is:
```
java -cp ./rapidwright-2023.2.1-standalone-lin64.jar com.xilinx.rapidwright.util.MakeBlackBox post_route.dcp shell.dcp bsv_userlogic_top_inst/bsvTopCore/sq
```

clavin-xlnx, 2024-03-12:
Thanks, I have reproduced the issue and will starting working on a fix once we root cause the error.

clavin-xlnx, 2024-03-12:
I have created a fix for the issue, you can find it in #970.  Let us know how it goes.

pwang7, 2024-03-13:
> I have created a fix for the issue, you can find it in #970. Let us know how it goes.

Great! Let me try it.

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx