Xilinx/RapidWright #968: Can't generate DeviceResources file for Versal

Quadrollopo, 2024-03-11:
I tried to generate the DeviceResources file for Versal with `rapidwright DeviceResourcesExample xcvc1902` 
but the output will always end with a NullPointerException:  
```==============================================================================
==                     Device Resources Dump: xcvc1902                      ==
==============================================================================
             Load Device:     1.405s    324.116MBs          |    627.945MBs (peak)
           populateEnums:     0.512s     44.979MBs          |    666.945MBs (peak)
Exception in thread "main" java.lang.NullPointerException
	at com.xilinx.rapidwright.interchange.DeviceResourcesWriter.writeAllSiteTypesToBuilder(DeviceResourcesWriter.java:620)
	at com.xilinx.rapidwright.interchange.DeviceResourcesWriter.writeDeviceResourcesFile(DeviceResourcesWriter.java:288)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:56)
	at com.xilinx.rapidwright.MainEntrypoint.main(MainEntrypoint.java:223)
```
I also tried different Versal devices but it will always end up like the previous output.  
For other non-Versal devices the command would work fine.

clavin-xlnx, 2024-03-11:
Thank you for reporting this issue.  I have created a workaround so that a Device Resources file can successfully be generated, you'll need this PR: https://github.com/Xilinx/RapidWright/pull/969.

A larger, more permanent fix will come a bit later.

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx