Xilinx/RapidWright #983: xcvh1782 wire/node mismatch

codepilot, 2024-04-05:
When iterating through all wires and nodes in a device, this code below should never output anything. It does for xcvh1782. It would be good to add to the tests.

```
for (Wire wire: node.getAllWiresInNode()) {
    if(!wire.getNode().equals(node)) {
        System.out.println(node.toString() + " != " + wire.toString() + ".getNode() " + wire.getNode().toString());
    }
}
```

codepilot, 2024-04-05:
[test-results.log](https://github.com/Xilinx/RapidWright/files/14879654/test-results.log)

Attached are the problematic entries

codepilot, 2024-04-05:
Not a vivado issue

```
get_wires -of_objects [get_nodes GTYP_QUAD_SINGLE_X0Y240/GTYP_QUAD_SITE_0_APB3PRDATA_0_]
```
results in:

GTYP_QUAD_SINGLE_X0Y240/GTYP_QUAD_SITE_0_APB3PRDATA_0_ CPM_G5_TILE_X0Y95/CPM5_ATOM_0_IFFCTRLQ3PRDATASF_0_

so CPM_G5_TILE_X0Y95/CPM5_ATOM_0_IFFCQ10TXONESZEROSFS is not in that list

This is the first line of the test-results.log

clavin-xlnx, 2024-04-08:
Thank you for reporting this issue!  I've diagnosed it to a missed check on the device creation process.  RapidWright attempts to compress the incoming device data and uses 16-bit storage for certain wire index representations.  The tile type `CPM_G5_TILE` in `xcvh1782` (and presumably other devices) has 79740 wires, clearly exceeding the 65536 threshold.  

We can probably rearrange the storage without much impact to memory utilization, but it will likely require a new set of device files and an updated release.  I will also add the appropriate checks and your test above to future release cycles.  

Are you able to work around this issue in the meantime, or is there an urgency in getting this resolved?

clavin-xlnx, 2024-04-15:
@codepilot - I've created PR #984 which should resolve this issue.  It requires new devices files and a new rapidwright-lib jar (both specified in the branch `device_widen_wire_index`.  Switching your local RapidWright git copy to that branch and running `./gradlew updateJars` should get you all the changes you need.  Give it a try and let us know how it goes.

codepilot, 2024-04-15:
> @codepilot - I've created PR #984 which should resolve this issue.  It requires new devices files and a new rapidwright-lib jar (both specified in the branch `device_widen_wire_index`.  Switching [...]

Will do

PR #1005 (2024-06-13): Test for wire/node mismatch reported in #983
Fixes #983
closed 2024-06-13 by clavin-xlnx