Xilinx/RapidWright #991: Running Rapidwright jython through script and eclipse setup issue

nandithaec, 2024-05-02:
I am trying the rapidwright jython using the interactive commandline format. The commands are working. But, when I put the commands in a script and run using `rapidwright jython script.py`, it gives the following error:

```
File "script.py", line 2, in <module>
    device = Device.getDevice("xczu9eg-ffvb1156-2-e")
NameError: name 'Device' is not defined
```

Next, although I did install eclipse, it does not seem to load the gitrepo

[image]

clavin-xlnx, 2024-05-02:
> I am trying the rapidwright jython using the interactive commandline format. The commands are working. But, when I put the commands in a script and run using `rapidwright jython script.py`, it gives [...]

The RapidWright Jython interpreter, as a convenience, will automatically import many classes automatically: (see https://github.com/Xilinx/RapidWright/blob/b9cdb4f6b40931237a196ba129f658612b9c5753/src/com/xilinx/rapidwright/util/Jython.java#L56).  When writing a script and running it, you will need to include those imports directly.  For example, in the error message mentioned, you can import the `Device` class by:
```
from com.xilinx.rapidwright.device import Device;
```

As for the issues setting up Eclipse, there could be a variety of issues going on.  Although it is a bit old, did you try following the Eclipse setup guide (https://www.rapidwright.io/docs/RapidWright_Eclipse_Setup.html)?   What version of Eclipse and what platform are you using?  What version of Git?

nandithaec, 2024-05-03:
Thanks. The first solution worked.

For the eclipse: Yes, I am following the same guidelines, does not work
yet. Let me try a bit more.
--
Regards,
Nanditha

closed 2025-05-13 by clavin-xlnx