Xilinx/RapidWright #992: Query on function arguments

nandithaec, 2024-05-14:
Continuing to try RapidWright through command line- interactive mode:
I am trying to do tab complete and trying out various commands. But, how do I know what should be the arguments. For example, how do I know what should be the argument into this count () 
>>> TileColumnPattern.count()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: count requires at least one argument

clavin-xlnx, 2024-05-14:
That is a good question.  Since RapidWright is written in Java, I don't believe the Javadoc strings are integrated into the tab-complete mechanism.  One way of accessing the Python docstrings is to run `help()` on the name of the method:
```
help(TileColumnPattern.count)
```
However, again because this is a Java method, it doesn't provide much.  Also, the `count()` method is actually a Python object method, it does not exist in the actual RapidWright code.  For APIs for which you are unfamiliar, I would check the Javadocs online https://www.rapidwright.io/javadoc/index.html

nandithaec, 2024-05-17:
Thank you. This documentation is useful. I have an additional question. Maybe it might be better to get on to a call briefly if possible? 
 I want to extract router or switch box related information for 5 different FPGA devices, to identify unique characteristics of the routing. The Router and RWRoute seem to be more for modifying routing, adding route etc. I did not find a method inside any class that gives router statistics of a device. Maybe I am missing something?

clavin-xlnx, 2024-05-20:
Here is some example code that will load three different devices and print out the wires and PIPs of that device's switch box:

In Jython/Python:
```python
from com.xilinx.rapidwright.device import Device
from com.xilinx.rapidwright.device import Tile
from com.xilinx.rapidwright.device import Wire
from com.xilinx.rapidwright.device import PIP
from com.xilinx.rapidwright.device import Node
from com.xilinx.rapidwright.device import TileTypeEnum

deviceNames = [Device.PYNQ_Z1, Device.KCU105, Device.AWS_F1]

for deviceName in deviceNames:
    device = Device.getDevice(deviceName)
    switchBox = device.getArbitraryTileOfType(TileTypeEnum.INT)
    if switchBox is None:
        # Series 7
        switchBox = device.getArbitraryTileOfType(TileTypeEnum.INT_L)
    print(device.getName() + ", Tile: " + str(switchBox))
    for wireIndex in range(0, switchBox.getWireCount()):
        wire = Wire(switchBox, wireIndex);
        print("  Wire: " + str(wireIndex) + " " + str(wire) + " " + str(wire.getIntentCode()))
        for pip in wire.getForwardPIPs():
            print("    PIP: " + str(pip))
    Device.releaseDeviceReferences();
```
The same functionality in Java:
```java
import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.PIP;
import com.xilinx.rapidwright.device.Tile;
import com.xilinx.rapidwright.device.TileTypeEnum;
import com.xilinx.rapidwright.device.Wire;

public class SwitchBoxInfo {

    public static void main(String[] args) {
        for (String deviceName : new String[] { Device.PYNQ_Z1, Device.KCU105, Device.AWS_F1 }) {
            Device device = Device.getDevice(deviceName);
            Tile switchBox = device.getArbitraryTileOfType(TileTypeEnum.INT);
            if (switchBox == null) {
                // Series 7
                switchBox = device.getArbitraryTileOfType(TileTypeEnum.INT_L);
            }
            System.out.println(device.getName() + ", Tile: " + switchBox);
            for (int wireIndex = 0; wireIndex < switchBox.getWireCount(); wireIndex++) {
                Wire wire = new Wire(switchBox, wireIndex);
                System.out.println("  Wire: " + wireIndex + " " + wire + " " + wire.getIntentCode());
                for (PIP pip : wire.getForwardPIPs()) {
                    System.out.println("    PIP: " + pip);
                }
            }
            Device.releaseDeviceReferences();
        }
    }
}
```

There are other APIs that can be used to explore the architecture, please see the [Javadocs](https://www.rapidwright.io/javadoc/index.html) for other options.  Let us know if you have other questions.

closed 2025-05-13 by clavin-xlnx