Xilinx/RapidWright #1010: Unexpected Laguna Sites Included in `design.getSiteInsts()`

RipperJ, 2024-06-20:
Hi. When I'm using `design.getSiteInsts()` and filter the laguna sites (SiteTypeEnum.LAGUNA) in it, I found some sites whose inner laguna registers are not used but included in `getSiteInsts()`. For example, as the following figure shows:

[image]

This site is included in `getSiteInsts()`, but whose property "IS_USED" in Vivado shows false.
Is this expected? If so, is it because that the wire connecting RXD4 is used?

clavin-xlnx, 2024-06-20:
Generally, those `SiteInst` objects should probably not be included.  Can you provide more information on how this design was created?  Are you able to share an example flow that causes these to appear?  We would be interested in tracking down the cause for their existence, thanks!

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx