Xilinx/RapidWright #1097: Working with JPickler

dg-deus, 2024-11-07:
Let's say I have data of dict type, and the data consists of various types including RapidWright objects like PIP.

And I tried to pickle/serialize the dict-type data, but failed.
```
from jpype.pickle import JPickler
with open('myData.pickle', 'wb') as f:
  JPickler(f).dump(my_dict)
```

The code fails with the following error:
`
java.lang.java.lang.InternalError: java.lang.InternalError: java.io.InvalidClassException: com.xilinx.rapidwright.device.PIP; imcompatible types for field a
`

RapidWright does not support JPickler?
Thank you in advance

clavin-xlnx, 2024-11-07:
Thanks for pointing this out, it looks like "Pickling of Java objects is restricted to classes that implement Serializable." according to the docs (https://jpype.readthedocs.io/en/latest/api.html?jpype-pickle-module=#jpype-pickle-module).  

It looks like most classes in RapidWright implement Serializable, but there are some that do not.  This inconsistency could be resolved in the next release as it is not a significant change to make.  However, if JPickler is using the Java Serializable interface, it might not produce very efficient results/performance.

clavin-xlnx, 2024-11-22:
I've attempted to update the rest of the RapidWright classes to implement Serializable, but I still can't get JPickler to work.  It is now complaining that `java.lang.Object` does not implement Serializable.  I'm not sure if there is any easy fix at this point.

Unfortunately, the `Serializable` interface in Java is not very performant and we don't typically use it.  

I tried using [Kryo ](https://github.com/EsotericSoftware/kryo)(which is bundled with RapidWright) as a Java serialization solution, but had issues with it as well as it requires all classes to have a no-arg constructor.  I'm afraid I don't have a good automatic solution for serializing Java objects at the moment, and there would certainly be some work to integrate it to work correctly with Python. 

Can you share a bit more about what you are trying to accomplish?  Perhaps we can help you better find an alternative solution.

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx