Xilinx/RapidWright #1114: Trying to run TimingAndWirelengthExample

alimousios98x, 2024-11-22:
I am trying to run TimingAndWirelengthExample  but it seems it isn't recognized as an application or any other method under rwroute. How can i build it or if it's something else I am doing wrong.

[image]

clavin-xlnx, 2024-11-22:
Perhaps you wanted to run this (as seen in [this tutorial](https://www.rapidwright.io/docs/ReportTimingExample.html))?
```
./gradlew run --args="ReportTimingExample <dcp file name>"
```

I see that we don't have `TimingAndWirelengthReport` as a listed execution target.  In that case, you can run it directly as long as you set the `CLASSPATH` environment variable like this (from your screenshot, it looks like you are using Windows PowerShell):

```
# Assuming PowerShell and you are located in the 'RapidWright' directory
$env:CLASSPATH = (Get-Location).Path + "\bin;" + (Get-Location).Path + "\jars\*"  
```

Then you can run the application directly:
```
java com.xilinx.rapidwright.rwroute.TimingAndWirelengthReport <input dcp file name>
```

clavin-xlnx, 2024-11-25:
Are you able to share the DCP you are using, or a DCP that can trigger this same behavior?  It would be much easier to debug this issue with an example test case.

alimousios98x, 2024-11-26:
Here is the one i used.

[test2_artix.zip](https://github.com/user-attachments/files/17910319/test2_artix.zip)

eddieh-xlnx, 2024-11-26:
Unfortunately, we do not have a timing model for the Series 7 family. Timing reports are only supported on UltraScale+ at this time.

closed 2024-12-05 by alimousios98x