Xilinx/RapidWright #1121: NetListCreation Example

alimousios98x, 2024-12-05:
I want to use the NetlistCreation example and incorporate a simple circuit and produce a checkpoint file but I am having issues with it. Is there any more detailed example or any other source on how to do this?

clavin-xlnx, 2024-12-05:
There is a very similar Jupyter Notebook that was created that uses similar elements to this example:
https://github.com/clavin-xlnx/RapidWright-binder/blob/4d658c450955d1cb9ba4989e68c020e88ad049ca/EDIFNetlistFromScratch.ipynb

You can try running it using mybinder (hit or miss, sometimes it won't work):
https://mybinder.org/v2/gh/clavin-xlnx/RapidWright-binder/python3-no-docker?labpath=HelloWorld.ipynb

If you can provide more details we can try helping you more with what you would like to accomplish.

alimousios98x, 2024-12-09:
I am trying to create the following circuit. I have trouble with making D-type flip-flops, how to make their input be inverted based on the clock edge and how to implement buffers as depicted.

[image]

clavin-xlnx, 2024-12-09:
Here is some Java code that will generate that circuit:
```
package com.xilinx.rapidwright.examples;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.NetType;
import com.xilinx.rapidwright.design.Unisim;
import com.xilinx.rapidwright.edif.EDIFCell;
import com.xilinx.rapidwright.edif.EDIFDirection;
import com.xilinx.rapidwright.edif.EDIFPort;
import com.xilinx.rapidwright.router.Router;

public class ExampleCircuit {

    private static final String CLK_NAME = "clk";

    private static final int CHAIN_LENGTH = 3;

    public static Cell createFDRE(Design d, String name, String loc) {
        Cell fdre = d.createAndPlaceCell(name, Unisim.FDRE, loc);
        Net clk = d.getNet(CLK_NAME);
        if (clk == null) {
            clk = d.createNet(CLK_NAME);
            EDIFPort topPortClk = d.getTopEDIFCell().createPort(CLK_NAME, EDIFDirection.INPUT, 1);
            clk.getLogicalNet().createPortInst(topPortClk);
        }
        clk.connect(fdre, "C");
        fdre.connectStaticSourceToPin(NetType.VCC, "CE");
        fdre.connectStaticSourceToPin(NetType.GND, "R");
        return fdre;
    }

    public static void main(String[] args) {
        String partName = "xc7a100tfgg676-2";

        // Create some common variables
        Design d = new Design("top",partName);
        d.setAutoIOBuffers(false);
        d.setDesignOutOfContext(true);
        EDIFCell top = d.getNetlist().getTopCell();

        Cell launch = createFDRE(d, "launch", "SLICE_X10Y10/AFF");
        Cell capture1 = createFDRE(d, "capture1", "SLICE_X10Y10/BFF");
        Cell capture2 = createFDRE(d, "capture2", "SLICE_X10Y10/CFF");
        
        Cell inv = d.createAndPlaceCell("inv", Unisim.LUT1, "SLICE_X10Y9/A6LUT");
        inv.addProperty("INIT", "2'h1"); // Inversion
        Net invOut = d.createNet("invOut");
        invOut.connect(launch, "D");
        invOut.connect(inv, "O");
        Net launchOut = d.createNet("launchOut");
        launchOut.connect(launch, "Q");
        launchOut.connect(capture2, "D");
        launchOut.connect(inv, "I0");
        
        // Create delay chain
        Cell[] delays = new Cell[CHAIN_LENGTH];
        Net[] delayOutputs = new Net[CHAIN_LENGTH];
        for (int i = 0; i < CHAIN_LENGTH; i++) {
            delays[i] = d.createAndPlaceCell("delay" + i, Unisim.LUT1, "SLICE_X10Y" +(11 + i) + "/A6LUT");
            delays[i].addProperty("INIT", "2'h0");
            delayOutputs[i] = d.createNet("delayOutput_" + i);
            delayOutputs[i].connect(delays[i], "O");
            if (i == 0) {
                launchOut.connect(delays[i], "I0");
            } else {
                delayOutputs[i - 1].connect(delays[i], "I0");
            }
        }
        delayOutputs[delays.length-1].connect(capture1, "D");
        
        Cell xor = d.createAndPlaceCell("xor", Unisim.XOR2, "SLICE_X11Y10/A6LUT");
        Net capture1Out = d.createNet("capture1Out");
        capture1Out.connect(capture1, "Q");
        capture1Out.connect(xor, "I0");
        Net capture2Out = d.createNet("capture2Out");
        capture2Out.connect(capture2, "Q");
        capture2Out.connect(xor, "I1");        
        
        Net alarm = d.createNet("alarm");
        alarm.connect(xor, "O");
        alarm.getLogicalNet().createPortInst(top.createPort("alarm", EDIFDirection.OUTPUT, 1));
        
        d.routeSites();
        new Router(d).routeDesign();

        d.writeCheckpoint(d.getName() + ".dcp");
    }
}

```
[image]

alimousios98x, 2024-12-10:
The warning regarding the clk source pin wont be disruptive for my task right? i want to use the merge function of rapidWright for this circuit with an existing dcp file. Also for the buffer which just acts as delay and doesn't change the value they just pass it, shouldn't I do 2'h2 instead of  delays[i].addProperty("INIT", "2'h0") which actually passes 0; 

[image]

[image]

clavin-xlnx, 2024-12-11:
> The warning regarding the clk source pin wont be disruptive for my task right? i want to use the merge function of rapidWright for this circuit with an existing dcp file. 

I didn't include circuitry for generating the clock so I just left it as an input.  You probably want to connect the clock from your top level design to it.  It might be easiest to use a blackbox in your top design and just use `read_checkpoint -cell` to populate it with this DCP.

> Also for the buffer which just acts as delay and doesn't change the value they just pass it, shouldn't I do 2'h2 instead of delays[i].addProperty("INIT", "2'h0") which actually passes 0;

Yes, you are correct, I put in the wrong INIT string, it should be `2'h2`.

alimousios98x, 2024-12-11:
Could you give me an example of how to do this with the clock?

clavin-xlnx, 2024-12-18:
> Could you give me an example of how to do this with the clock and then merge it with another DCP.

My apologies for the late reply.  I don't have a design targeting the xc7a100t, but I do have one for the xcku040 that I will use as an example.  Please download the following tutorial design by running:

```
wget http://www.rapidwright.io/docs/_downloads/kcu105_step2.zip
unzip kcu105_step2.zip
```

Then, run this modified version of the code above (retargeted for the xcku040) that combines the design in `kcu105/kcu_105_routed.dcp` with the circuit created and connects up the `clk` input:

```java
package com.xilinx.rapidwright.examples;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Module;
import com.xilinx.rapidwright.design.ModuleInst;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.NetType;
import com.xilinx.rapidwright.design.Unisim;
import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.Site;
import com.xilinx.rapidwright.edif.EDIFCell;
import com.xilinx.rapidwright.edif.EDIFDirection;
import com.xilinx.rapidwright.edif.EDIFNet;
import com.xilinx.rapidwright.edif.EDIFPort;
import com.xilinx.rapidwright.router.Router;

public class ExampleCircuit {

    private static final String CLK_NAME = "clk";

    private static final int CHAIN_LENGTH = 3;

    public static Cell createFDRE(Design d, String name, String loc) {
        Cell fdre = d.createAndPlaceCell(name, Unisim.FDRE, loc);
        Net clk = d.getNet(CLK_NAME);
        if (clk == null) {
            clk = d.createNet(CLK_NAME);
            EDIFPort topPortClk = d.getTopEDIFCell().createPort(CLK_NAME, EDIFDirection.INPUT, 1);
            clk.getLogicalNet().createPortInst(topPortClk);
        }
        clk.connect(fdre, "C");
        fdre.connectStaticSourceToPin(NetType.VCC, "CE");
        fdre.connectStaticSourceToPin(NetType.GND, "R");
        return fdre;
    }

    public static void main(String[] args) {
        String partName = Device.KCU105;

        // Create some common variables
        Design d = new Design("circuit", partName);
        d.setAutoIOBuffers(false);
        d.setDesignOutOfContext(true);
        EDIFCell top = d.getNetlist().getTopCell();

        Site ref = d.getDevice().getSite("SLICE_X52Y191");
        Cell launch = createFDRE(d, "launch", ref + "/AFF");
        Cell capture1 = createFDRE(d, "capture1", ref + "/BFF");
        Cell capture2 = createFDRE(d, "capture2", ref + "/CFF");

        Cell inv = d.createAndPlaceCell("inv", Unisim.LUT1,
                ref.getNeighborSite(0, -1).getName() + "/A6LUT");
        inv.addProperty("INIT", "2'h1"); // Inversion
        Net invOut = d.createNet("invOut");
        invOut.connect(launch, "D");
        invOut.connect(inv, "O");
        Net launchOut = d.createNet("launchOut");
        launchOut.connect(launch, "Q");
        launchOut.connect(capture2, "D");
        launchOut.connect(inv, "I0");

        // Create delay chain
        Cell[] delays = new Cell[CHAIN_LENGTH];
        Net[] delayOutputs = new Net[CHAIN_LENGTH];
        for (int i = 0; i < CHAIN_LENGTH; i++) {
            delays[i] = d.createAndPlaceCell("delay" + i, Unisim.LUT1,
                    ref.getNeighborSite(0, i) + "/A6LUT");
            delays[i].addProperty("INIT", "2'h0");
            delayOutputs[i] = d.createNet("delayOutput_" + i);
            delayOutputs[i].connect(delays[i], "O");
            if (i == 0) {
                launchOut.connect(delays[i], "I0");
            } else {
                delayOutputs[i - 1].connect(delays[i], "I0");
            }
        }
        delayOutputs[delays.length - 1].connect(capture1, "D");

        Cell xor = d.createAndPlaceCell("xor", Unisim.XOR2, ref.getNeighborSite(1, 0) + "/A6LUT");
        Net capture1Out = d.createNet("capture1Out");
        capture1Out.connect(capture1, "Q");
        capture1Out.connect(xor, "I0");
        Net capture2Out = d.createNet("capture2Out");
        capture2Out.connect(capture2, "Q");
        capture2Out.connect(xor, "I1");

        Net alarm = d.createNet("alarm");
        alarm.connect(xor, "O");
        alarm.getLogicalNet().createPortInst(top.createPort("alarm", EDIFDirection.OUTPUT, 1));

        d.routeSites();
        new Router(d).routeDesign();

        // d.writeCheckpoint(d.getName() + ".dcp");

        // Load a top-level design
        Design topDesign = Design.readCheckpoint("kcu105/kcu105_route.dcp");

        // Instantiate the created circuit inside our topDesign just loaded
        ModuleInst mi = topDesign.createModuleInst(d.getName(), new Module(d));
        mi.placeOnOriginalAnchor();

        // Connect up the clock
        EDIFNet clkNet = topDesign.getTopEDIFCell().getNet("sys_clk");
        clkNet.createPortInst("clk", mi.getCellInst());

        topDesign.writeCheckpoint("combined.dcp");

    }
}
```
Note that I chose to place the circuit far away from all the other implemented logic just to avoid any collisions.
Then, you can load `combined.dcp` in Vivado and route it to route up the clock and VCC/GND inputs to the circuit, in Vivado run:

```
open_checkpoint combined.dcp
route_design
```
[image]

alimousios98x, 2025-01-08:
Hello i was wondering is there a way to create a full project (xpr file) from this dcp file that i generated from the jupyter notebook in order to be able to edit it from vivado afterwards so its not depended on your libraries

alimousios98x, 2025-01-08:
Basically i have an existing project and i want to make a a second porject ( the one you helped me write the small circuit) and be able to further edit it in vivado as a standalone project and then  merge them ( i think i can use the rapidwright merge for that)

clavin-xlnx, 2025-01-08:
> Hello i was wondering is there a way to create a full project (xpr file) from this dcp file that i generated from the jupyter notebook in order to be able to edit it from vivado afterwards so its not depended on your libraries

One idea is you can load the DCP into Vivado and then use `write_verilog` to write out the netlist and use that as a starting point in a Vivado project.  This won't save the placement/routing though, you would have to export that separately, I believe.

> Basically i have an existing project and i want to make a a second porject ( the one you helped me write the small circuit) and be able to further edit the dcp file in vivado as a standalone project and then merge them ( i think i can use the rapidwright merge for that).

You can use the merge to combine two independent designs, but if you want them to interact, it may not always match what you want.  You can either use RapidWright or Vivado Tcl to instantiate black boxes and populate them with a DCP or wire up connections using RapidWright APIs and/or Vivado Tcl.

closed 2025-01-15 by alimousios98x