Xilinx/RapidWright #1123: Unable to modify Versal (VCK190) DCP.

Arjun-Khanna, 2024-12-09:
I'm using the latest version of RapidWright (2024.2). However it is unable to generate the intersite_delay_terms.txt file for Versal. The error is as follows:
An error occurred: java.lang.RuntimeException: Error reading file:timing/versal/intersite_delay_terms.txt

clavin-xlnx, 2024-12-09:
Thank you for the feedback.  Unfortunately, we don't have timing model numbers for Versal yet in RapidWright.  We do have them for UltraScale+ devices.  We do have support in RWRoute (the RapidWright router) to route in Versal.  Let us know how we can best help with your application.

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx