Xilinx/RapidWright #1136: Place and Route

alimousios98x, 2025-01-11:
I have an existing project where i have set some specific areas as do not touch from vivado. In a previous issues you helped create with a rapidWright jupyter notbook a dcp file where i created a small circuit in specific specific area of the device ( the one that vivado wont touch). Is there a way for me to use some kind of method of rapidWright that can place and route the small circuit dcp file on the dcp file of my project or some other way to achieve this ( I have the custom circuit dcp file and my vivado project). I cannot use the merge method cause this project is for a non ultra scale device unfortunately.

clavin-xlnx, 2025-01-13:
Probably the simplest way is for you to add a [black box](https://docs.amd.com/r/2022.1-English/ug912-vivado-properties/BLACK_BOX) to your project design such that the black boxed module matches the interface of your RapidWright-generated circuit.  Then, you can populate the black box using `read_checkpoint -cell <blackbox_inst_name>  <RapidWright-design.dcp>`. This will join the two designs together.

alimousios98x, 2025-01-13:
I will try that. Could i also use your jupyter and instead of placing and routing my circuit in a new dcp to do it at the existing dcp of  my project? Therefore I uplaod my dcp project locally to jupyter and load it.

clavin-xlnx, 2025-01-13:
> I will try that. Could i also use your jupyter and instead of placing and routing my circuit in a new dcp to do it at the existing dcp of my project? Therefore I uplaod my dcp project locally to jupyter and load it.

I'm not sure if I fully understand your question, but RapidWright cannot read/write Vivado projects.  It only reads and writes design checkpoint files.

alimousios98x, 2025-01-13:
Yeah so basically can I place and route my custom circuit onto the dcp of my vivado project via Jupyter RapidWright.

alimousios98x, 2025-01-14:
I am trying to place and route an extra logic on an existing dcp of my vivado project as u have shown me before in a previous issue but it doesnt seem to match the device with the dcp.
 
[image]

The device i am using is a chip whisperer with this part so shouldn't it work by calling the device part or it needs the device name? ( you had a dcp for KCU105 and you did this "String partName = Device.KCU105;", where as i am referencing the part )
[image]

[image]

The dcp file:
[dcpFile.zip](https://github.com/user-attachments/files/18410713/dcpFile.zip)

Here is my code:
```
package com.xilinx.rapidwright.examples;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Module;
import com.xilinx.rapidwright.design.ModuleInst;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.NetType;
import com.xilinx.rapidwright.design.Unisim;
import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.Site;
import com.xilinx.rapidwright.edif.EDIFCell;
import com.xilinx.rapidwright.edif.EDIFDirection;
import com.xilinx.rapidwright.edif.EDIFNet;
import com.xilinx.rapidwright.edif.EDIFPort;
import com.xilinx.rapidwright.router.Router;

public class circuit1 {

    private static final String CLK_NAME = "clk";

    private static final int CHAIN_LENGTH = 3;

    public static Cell createFDRE(Design d, String name, String loc) {
        Cell fdre = d.createAndPlaceCell(name, Unisim.FDRE, loc);
        Net clk = d.getNet(CLK_NAME);
        if (clk == null) {
            clk = d.createNet(CLK_NAME);
            EDIFPort topPortClk = d.getTopEDIFCell().createPort(CLK_NAME, EDIFDirection.INPUT, 1);
            clk.getLogicalNet().createPortInst(topPortClk);
        }
        clk.connect(fdre, "C");
        fdre.connectStaticSourceToPin(NetType.VCC, "CE");
        fdre.connectStaticSourceToPin(NetType.GND, "R");
        return fdre;
    }

    public static void main(String[] args) {
        String partName = "xc7a100t";

        // Create some common variables
        Design d = new Design("circuit", partName);
        d.setAutoIOBuffers(false);
        d.setDesignOutOfContext(true);
        EDIFCell top = d.getNetlist().getTopCell();

        Site ref = d.getDevice().getSite("SLICE_X48Y79");
        Cell launch = createFDRE(d, "launch", ref + "/AFF");
        Cell capture1 = createFDRE(d, "capture1", ref + "/BFF");
        Cell capture2 = createFDRE(d, "capture2", ref + "/CFF");

        Cell inv = d.createAndPlaceCell("inv", Unisim.LUT1,
                ref.getNeighborSite(0, -1).getName() + "/A6LUT");
        inv.addProperty("INIT", "2'h1"); // Inversion
        Net invOut = d.createNet("invOut");
        invOut.connect(launch, "D");
        invOut.connect(inv, "O");
        Net launchOut = d.createNet("launchOut");
        launchOut.connect(launch, "Q");
        launchOut.connect(capture2, "D");
        launchOut.connect(inv, "I0");

        // Create delay chain
        Cell[] delays = new Cell[CHAIN_LENGTH];
        Net[] delayOutputs = new Net[CHAIN_LENGTH];
        for (int i = 0; i < CHAIN_LENGTH; i++) {
            delays[i] = d.createAndPlaceCell("delay" + i, Unisim.LUT1,
                    ref.getNeighborSite(0, i) + "/A6LUT");
            delays[i].addProperty("INIT", "2'h0");
            delayOutputs[i] = d.createNet("delayOutput_" + i);
            delayOutputs[i].connect(delays[i], "O");
            if (i == 0) {
                launchOut.connect(delays[i], "I0");
            } else {
                delayOutputs[i - 1].connect(delays[i], "I0");
            }
        }
        delayOutputs[delays.length - 1].connect(capture1, "D");

        Cell xor = d.createAndPlaceCell("xor", Unisim.XOR2, ref.getNeighborSite(1, 0) + "/A6LUT");
        Net capture1Out = d.createNet("capture1Out");
        capture1Out.connect(capture1, "Q");
        capture1Out.connect(xor, "I0");
        Net capture2Out = d.createNet("capture2Out");
        capture2Out.connect(capture2, "Q");
        capture2Out.connect(xor, "I1");

        Net alarm = d.createNet("alarm");
        alarm.connect(xor, "O");
        alarm.getLogicalNet().createPortInst(top.createPort("alarm", EDIFDirection.OUTPUT, 1));

        d.routeSites();
        new Router(d).routeDesign();

        // d.writeCheckpoint(d.getName() + ".dcp");

        // Load a top-level design
        Design topDesign = Design.readCheckpoint("D:/ALEXANDROS/Documents/PROJECT/designAES.dcp");

        // Instantiate the created circuit inside our topDesign just loaded
        ModuleInst mi = topDesign.createModuleInst(d.getName(), new Module(d));
        mi.placeOnOriginalAnchor();

        // Connect up the clock
        EDIFNet clkNet = topDesign.getTopEDIFCell().getNet("sys_clk");
        clkNet.createPortInst("clk", mi.getCellInst());

        topDesign.writeCheckpoint("combined.dcp");

    }
}
```

clavin-xlnx, 2025-01-14:
It looks like you are trying to target the `xc7a100tftg256-2`, this is the part name that you'll want in your code.  I opened the `designAES.dcp` however, and it is targeting the `xc7a35tftg256-2` which is a different part.  Since these two don't match, you are getting the error from RapidWright.  You'll either need to change the RapidWright code to match the DCP or rebuild the DCP onto the `xc7a100tftg256-2` part.

alimousios98x, 2025-01-14:
I have tried adding the full part name as you mentioned but the same error occurs.
[image]

alimousios98x, 2025-01-15:
You are correct, thank you. For fully routing it with the clock that is being used for the design i should replace the "sys_clk" with the name of the clock that is being used? What if i see 3 clocks being used in the design.
```
        // Connect up the clock
       EDIFNet clkNet = topDesign.getTopEDIFCell().getNet("sys_clk");
       clkNet.createPortInst("clk", mi.getCellInst());
```

[image]

clavin-xlnx, 2025-01-15:
> You are correct, thank you. For fully routing it with the clock that is being used for the design i should replace the "sys_clk" with the name of the clock that is being used? What if i see 3 clocks being used in the design.

You will need to decide which clock you want the circuit connected to based on its attributes, period and relationship to the rest of the design.  Be sure to get the name of the clock being driven by the BUFG such as `crypt_clk` or `usb_clk_buf`.  The code you have expects the clock net to be at the top level of the top cell.  If the clock you want to connect to is inside a level of hierarchy, you'll need to create new ports/nets to bring the clock out to the circuit.

alimousios98x, 2025-01-16:
All the clocks of the project have the same status INTRASITE meaning that they aren't on the top level? Lets say I want the pll_clk1 so I need to to create new ports/nets to bring the clock out to the circuit and how would I do that if you could assist? Or is there a way to ensure that the clock signal (pll_clk1) is routed globally through a BUFG?

[image]

[image]

alimousios98x, 2025-01-17:
It seems that after i route the design the added circuit is successfully routed to the clock mentioned (crypt_clk) but it messes with some others nets for some reason, what could be the reason for that?

Before **route_design**:

[image]

After **route_design**:

[image]

I attached the dcp file with which if you do **route_design** the problem will occur.

[COMB1.zip](https://github.com/user-attachments/files/18453696/COMB1.zip)

clavin-xlnx, 2025-01-17:
It looks like the LUT1 cell `circuit/inv` has its cell pin mappings incorrectly set.  Perhaps Vivado made a change/swap.  You can work around it by unplacing the cell and then replacing it again:

```
unplace_cell  [get_cells [list  circuit/inv] -filter {((is_primitive==true && primitive_level!="INTERNAL")  && (loc!=""))}]
place_cell circuit/inv SLICE_X48Y80/A6LUT
route_design
```

When I do this I get a completely routed solution without errors:
```
report_route_status
Design Route Status
                                               :      # nets :
   ------------------------------------------- : ----------- :
   # of logical nets.......................... :        1398 :
       # of nets not needing routing.......... :         440 :
           # of internally routed nets........ :         439 :
           # of nets with no loads............ :           1 :
       # of routable nets..................... :         958 :
           # of fully routed nets............. :         958 :
       # of nets with routing errors.......... :           0 :
   ------------------------------------------- : ----------- :
```

alimousios98x, 2025-01-18:
The design is fully routed after fixing the wrong mapping you mentioned. My goal is to match the critical path (or WNS) of the custom circuit to the critical path of the original design. I have the dcp of the custom design on its own as I created it from your Jupyter. Is there a way via RapidWright to get reports regarding the timing and its critical path (it's not for an ultrascale device). Or is there any other way to extract this information from a dcp file?

clavin-xlnx, 2025-01-21:
> When i do timing report shouldn't I see a path that is involved with this new added circuit?

Not necessarily.  If the added circuit doesn't have any critical paths relative to the main circuit, they wouldn't show up in the list.  Generally, these timing reports just show the paths with the worst negative slack.

alimousios98x, 2025-01-22:
Is there another relative RapidWright tool or Vivado that shows me timing information for example my given custom circuit in the total design or standalone which can also be in a form a dcp as its own?

clavin-xlnx, 2025-01-22:
> Is there another relative RapidWright tool or Vivado that shows me timing information for example my given custom circuit in the total design or standalone which can also be in a form a dcp as its own?

You can get timing information by loading the design in context in Vivado.  You can request specific paths be timed by Vivado if you want that information.  Try using `get_timing_paths` (see https://docs.amd.com/r/en-US/ug835-vivado-tcl-commands/get_timing_paths).

alimousios98x, 2025-01-28:
Regarding my alarm how would I go about to connect it to an output port in order to connect it to my oscillator ? Let's assume that I have found a port, do I need to instantiate it somehow or activate it before I connect a net between it and the alarm?

clavin-xlnx, 2025-01-28:
> Regarding my alarm how would I go about to connect it to an output port in order to connect it to my oscillator ? Let's assume that I have found a port, do I need to instantiate it somehow or activate it before I connect a net between it and the alarm?

Yes, you would need to instantiate the IO and place it on the package pin you've selected you would use `create_cell`.  Next, you would need to connect it to the net of interest using `connect_net`.

alimousios98x, 2025-02-18:
If for example i want to connect my alarm output to package pin D15 how would I do that with RW cause I havent figured it out yet? Do i have to assign that to a top-level port my FPGA design via vivado prior or I can do this with RW somehow?

closed 2025-02-18 by alimousios98x