Xilinx/RapidWright #1138: Retrieving SitePinInst from Logical Cell Pin via a Route-Thru LUT

hcook40, 2025-01-22:
Hello,

I have a placed and routed design where I am trying to get the SitePinInst from a cell's logical pin. The site is already routed. The site net in question uses a LUT6 as a route-thru LUT. The image below shows the internal site net in Vivado:

[image]

My first attempt used `cell.unrouteLogicalPinInSite()`, but that returned null. My second attempt tried to manually iterate through used sitePIPs to get to the sitePin, but that also failed.

Further investigation shows that the route-thru LUTs are not instantiated as sitePIPs (although there are sitePIPs that correspond to route-thru LUTs). Instead, RapidWright sees route-thru LUTs as "dummy" cells that have the same name as the sink cell. Unfortunately, I cannot find a good way to associate these dummy cells with the cell pin in question. Is there a method for this? Alternatively, is there a way to convert the dummy cells into sitePIPs?

Here is my code and corresponding checkpoint files:

```java
package com.xilinx.rapidwright.examples;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.SiteInst;
import com.xilinx.rapidwright.design.SitePinInst;
import com.xilinx.rapidwright.device.BELPin;
import com.xilinx.rapidwright.device.SitePIP;
import com.xilinx.rapidwright.device.SitePin;

import java.util.Collection;
import java.util.List;

public class GetSitePinInstFromCellPin {
    public static void main(String[] args) {
        final String DCP_FILE = "bgm.dcp";
        final String CELL_NAME = "bgm/a2_add/u1/_inferred__0/i__carry";
        final String CELL_PIN_NAME = "S[0]";

        Design design = Design.readCheckpoint(DCP_FILE, DCP_FILE.replace(".dcp", ".edf"));

        Cell cell = design.getCell(CELL_NAME);
        SitePinInst sitePinInst = cell.unrouteLogicalPinInSite(CELL_PIN_NAME); // returns null
        System.out.println(sitePinInst);

        BELPin belPin = cell.getBEL().getPin(cell.getPhysicalPinMapping(CELL_PIN_NAME)); // get BEL pin from logical cell pin
//        String sitePinName = belPin.getConnectedSitePinName();

        SiteInst siteInst = cell.getSiteInst();

//        //infinite loop
//        while (!belPin.isDedicatedSitePin()) {
//            SitePIP sitePIP = siteInst.getUsedSitePIP(belPin.getBELName());
//            if (sitePIP != null) {
//                belPin = sitePIP.getInputPin();
//            }
//            else {
//                // If there are no pips, we can get the source BEL pin directly
//                belPin = belPin.getSourcePin();
//            }
//        }
//        SitePin sitePin = belPin.getSitePin(siteInst.getSite());
//        System.out.println(sitePin);

        // check if LUT route-thru pips are used
        List<SitePIP> usedSitePips = siteInst.getUsedSitePIPs();
        System.out.println(usedSitePips); // Doesn't contain LUT route-thru PIPs

        // check for route-thru LUTs as Cells
        Collection<Cell> siteCells = siteInst.getCells();
        System.out.println(siteCells); // 3 cells with the same name, "bgm/a2_add/u1/_inferred__0/i__carry". One is a carry4, the other two are route-thru LUTs
    }
}

```

[bgm.zip](https://github.com/user-attachments/files/18507096/bgm.zip)

clavin-xlnx, 2025-01-22:
We have an API for getting the `SitePinInst` from a `Cell`'s logical pin [`Cell.getSitePinFromLogicalPin()`](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Cell.html#getSitePinFromLogicalPin-java.lang.String-java.util.List-):

I added the following line right after calling `getCell()`:
```
        Cell cell = design.getCell(CELL_NAME);
        System.out.println(cell.getSitePinFromLogicalPin(CELL_PIN_NAME, null));
```
And the output is the attached `SitePinInst`:
```
IN SLICE_X31Y26.A5
```

hcook40, 2025-01-22:
That's exactly what I needed. I don't know how I missed that function. Thanks!

closed 2025-01-22 by hcook40