Xilinx/RapidWright #1143: Inquiry about RWRoute support  Series7 series

liibin, 2025-02-08:
I noticed in the documentation that RWRoute currently only supports UltraScale+ devices. I have two related questions:
1.Why was UltraScale+ chosen as the initial supported architecture?
2.Is there a roadmap to extend support for Series7 devices (e.g., Virtex-7, Kintex-7)?Is it feasible to modify RWRoute to support the Series7 series？

clavin-xlnx, 2025-02-08:
> 1.Why was UltraScale+ chosen as the initial supported architecture? 

At the time of its development, UltraScale+ was the mainstream target architecture and since each architecture is a bit different, we decided to focus on just UltraScale+.   Since then, preliminary support has been added for UltraScale and Versal.

> 2.Is there a roadmap to extend support for Series7 devices (e.g., Virtex-7, Kintex-7)?Is it feasible to modify RWRoute to support the Series7 series？

Although we don't have Series 7 explicitly on the roadmap, it would be a nice addition to RWRoute.  It is certainly feasible to do so, it would just take someone a bit of effort to create case handling for where the Series 7 architecture differs from others.  We are open to supporting external efforts to do so.  Please let us know what you are trying to accomplish and perhaps we can explore options to help you accomplish your goals!

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx