Xilinx/RapidWright #1150: get_nets different with Vivado

coherent17, 2025-02-21:
Hi Rapidwright Expert,
As the Java document mentioned, the get_nets in rapidwright corresponding to the get_nets -hier in Vivado, but the number of the nets are different:

Vivado:

llength [get_nets -hier]
-> 4740034

rapidwright:
design.getNets.size()
-> 1206612

Is there any suggestion or recommendation on this issue?

Thanks and regards!

Coherent

clavin-xlnx, 2025-02-21:
To get the number of hierarchical nets, you'll need to get the nets from the logical netlist:

```java
Design d = Design.readCheckpoint("your.dcp");
System.out.println(d.getNetlist().getParentNetMap().size() + 2); // we need to add 2 for VCC and GND
````

`Design.getNets().size()` gets the number of physical (routable) nets which is generally smaller as it doesn't account for hierarchy.

coherent17, 2025-02-21:
Hi,

Thanks for help! The number are matching now!

Regards!

Coherent

coherent17, 2025-02-21:
Hi, I have another follow up question for this, is there a tcl command in Vivado that get the physical (routable) nets that design.getNets() returned?

clavin-xlnx, 2025-02-21:
> Hi, I have another follow up question for this, is there a tcl command in Vivado that get the physical (routable) nets that design.getNets() returned?

I'm not sure of an exact method, but `report_route_status` reports "logical nets" (first entry in the table) which is pretty close.

coherent17, 2025-02-22:
Hi, thanks for your useful insight, I may close this issue as complete!

Regards!

closed 2025-02-22 by coherent17