Xilinx/RapidWright #1153: how to place a RAMB18E1 EDIFHierCellInst onto a FIFO18E1 Site?

TotoroTron, 2025-03-08:
Hi RW team, I am building a packer + placer as part of my MS grad project and have so far built a crude but working simulated annealer that moves/swaps on the SiteInst/Site level for the 7-Series arch. I think I can improve the placement quality by taking advantage of SiteInst->Site compatibility between RAMB18E1 and FIFO18E1 but I can't figure out how to do that in RW.

In my packer I have this function:
```java
    private List<SiteInst> packRAMSiteInsts(List<EDIFHierCellInst> RAMCells)
            throws IOException {
        List<SiteInst> RAMSiteInsts = new ArrayList<>();
        writer.write("\n\nPacking RAMBCells... (" + RAMCells.size() + ")");
        for (EDIFHierCellInst ehci : RAMCells) {
            Site selectedSite = selectRAMSite();
            SiteTypeEnum ste = selectedSite.getSiteTypeEnum();
            SiteInst si = null;
            if (ste == SiteTypeEnum.RAMB18E1) {
                si = new SiteInst(
                        ehci.getFullHierarchicalInstName(), design, SiteTypeEnum.RAMB18E1, selectedSite);
                si.createCell(ehci, si.getBEL("RAMB18E1"));
            }
            if (ste == SiteTypeEnum.FIFO18E1) {
                si = new SiteInst(
                        ehci.getFullHierarchicalInstName(), design, SiteTypeEnum.FIFO18E1, selectedSite);
                si.createCell(ehci, si.getBEL("RAMB18E1"));
                System.out.println("SiteInst:" + si);
                System.out.println("\tCell-BEL Map:");
                for (Map.Entry<String, Cell> entry : si.getCellMap().entrySet()) {
                    System.out.println("\t\t<" + entry.getKey() + ", " + entry.getValue() + ">");
                }
            }
            si.routeSite();
            RAMSiteInsts.add(si);
        }
        return RAMSiteInsts;
    } // end packRAMSiteInsts()
```

where selectRAMSite() is:
```java
    private Site selectRAMSite() {
        List<Site> compatibleSites = new ArrayList<>();
        compatibleSites.addAll(availableSites.get(SiteTypeEnum.RAMB18E1));
        compatibleSites.addAll(availableSites.get(SiteTypeEnum.FIFO18E1));
        if (compatibleSites.isEmpty()) {
            throw new IllegalStateException(
                    "ERROR: device or clock region contains no Sites of type FIFO18E1 or RAMB18E1!");
        }
        compatibleSites.sort(
                Comparator.comparingInt(Site::getInstanceX)
                        .thenComparingInt(Site::getInstanceY)
                        .reversed());
        Site selectedSite = compatibleSites.get(0);
        SiteTypeEnum ste = selectedSite.getSiteTypeEnum();

        availableSites.get(ste).remove(selectedSite);
        occupiedSites.get(ste).add(selectedSite);
        return selectedSite;
    }

```
where availableSites and occupiedSites are just bookkeeping maps. (the placer has no concept of global placement or legalization):
```java
    protected Map<SiteTypeEnum, List<Site>> occupiedSites;
    protected Map<SiteTypeEnum, List<Site>> availableSites;
```

The System.out.println() statements in packRAMSiteInsts() come out like the following:
```
SiteInst:SiteInst(name="fir_filter_inst/genblk3[11].xpm_memory_spram_inst/xpm_memory_base_inst/gen_wr_a.gen_word_narrow.mem_reg", type="FIFO18E1", site="RAMB18_X5Y6")
        Cell-BEL Map:
SiteInst:SiteInst(name="fir_filter_inst/genblk3[29].xpm_memory_spram_inst/xpm_memory_base_inst/gen_wr_a.gen_word_narrow.mem_reg", type="FIFO18E1", site="RAMB18_X5Y4")
        Cell-BEL Map:
SiteInst:SiteInst(name="fir_filter_inst/genblk2[19].xpm_memory_sprom_inst/xpm_memory_base_inst/gen_rd_a.gen_rd_a_synth_template.gen_rf_narrow_reg.douta_reg_reg", type="FIFO18E1", site="RAMB18_X5Y2")
        Cell-BEL Map:
SiteInst:SiteInst(name="fir_filter_inst/genblk2[6].xpm_memory_sprom_inst/xpm_memory_base_inst/gen_rd_a.gen_rd_a_synth_template.gen_rf_narrow_reg.douta_reg_reg", type="FIFO18E1", site="RAMB18_X5Y0")
        Cell-BEL Map:
SiteInst:SiteInst(name="fir_filter_inst/genblk3[30].xpm_memory_spram_inst/xpm_memory_base_inst/gen_wr_a.gen_word_narrow.mem_reg", type="FIFO18E1", site="RAMB18_X4Y58")
        Cell-BEL Map:
SiteInst:SiteInst(name="fir_filter_inst/genblk3[26].xpm_memory_spram_inst/xpm_memory_base_inst/gen_wr_a.gen_word_narrow.mem_reg", type="FIFO18E1", site="RAMB18_X4Y56")
        Cell-BEL Map:

etc...
```
It seems I am not creating the Cell in the SiteInst correctly and the packer simply skips the cell. The RAMB18E1 cells that are assigned to RAMB18E1 Sites are processed and placed just fine, just not the ones assigned to FIFO Sites. Do the EHCI's themselves have to be of FIFO type or something?

In another one of my attempts, I tried the following:
```java
           if (ste == SiteTypeEnum.FIFO18E1) {
                si = new SiteInst(
                        ehci.getFullHierarchicalInstName(), design, SiteTypeEnum.FIFO18E1, selectedSite);
                si.createCell(ehci, si.getBEL("RAMB18E1"));
```
I get this error:
```
Exception in thread "main" java.lang.RuntimeException: ERROR: Site type FIFO18E1 not supported for cell type RAMB18E1
        at com.xilinx.rapidwright.design.h.a(Unknown Source)
        at com.xilinx.rapidwright.design.h.a(Unknown Source)
        at com.xilinx.rapidwright.design.h.a(Unknown Source)
        at com.xilinx.rapidwright.design.Design.placeCell(Unknown Source)
        at com.xilinx.rapidwright.design.SiteInst.createCell(Unknown Source)
        at placer.PackerBasic1.packRAMSiteInsts(PackerBasic1.java:225)
        at placer.PackerBasic1.packDesign(PackerBasic1.java:106)
        at placer.Packer.run(Packer.java:62)
        at placer.Main.main(Main.java:55)
```

clavin-xlnx, 2025-03-10:
Great question!  It turns out that `Site` objects in Vivado can take on multiple site types.  For a given BRAM tile in Series 7, there are 3 sites, for example, in a `xc7z010clg400-1`, the BRAM tile `BRAM_L_X4Y35` has the sites:

[image]

Notice that there are 6 rows, but only 3 unique entries under name.  This is because there are site types that are primary and secondary.  Primary site types are the default type, for example, `RAMB18_X0Y14` has a primary type of `FIFO18E1`:

[image]

But as the image above shows in Vivado, the site can also be configured to be of type `RAMB18E1`.  I think this is where you might be having trouble as attempting to place a `RAMB18E`1` cell onto a site configured as type `FIFO18E1` is not allowed.  We need to change the site type to `RAMB18E1` first before attempting to place a cell there:

[image]

To do that in RapidWright, we simply need to change the `SiteTypeEnum` of a `SiteInst` to the desired type.  So in this example, we would create a new SiteInst with `SiteTypeEnum.RAMB18E1` instead of `SiteTypeEnum.FIFO18E1`.  Modifying your code above, it should look something like this:
```java
// ...
            if (ste == SiteTypeEnum.FIFO18E1) {
                si = new SiteInst(
                        ehci.getFullHierarchicalInstName(), design, SiteTypeEnum.RAMB18E1/*instead of FIFO18E1*/, selectedSite);
// ...
```

Let us know if you run into any more issues.

TotoroTron, 2025-03-10:
Hi clavin, thanks for the in-depth response. You're right, the SiteInst constructor's SiteTypeEnum should be RAMB18E1, and si.createCell() should get BEL RAMB18E1 too. I wrongly assumed that separate handling was required for RAMB18E1 and FIFO18E1 Sites, but it turns out the below function sufficiently handles both:

```java
    private List<SiteInst> packRAMSiteInsts(List<EDIFHierCellInst> RAMCells)
            throws IOException {
        List<SiteInst> RAMSiteInsts = new ArrayList<>();
        writer.write("\n\nPacking RAMBCells... (" + RAMCells.size() + ")");
        for (EDIFHierCellInst ehci : RAMCells) {
            Site selectedSite = selectRAMSite(); // selects uniformly randomly between RAMB18E1 and FIFO18E1 Site
            SiteInst si = new SiteInst(
                    ehci.getFullHierarchicalInstName(), design, SiteTypeEnum.RAMB18E1, selectedSite);
            si.createCell(ehci, si.getBEL("RAMB18E1"));
            si.routeSite();
            RAMSiteInsts.add(si);
        }
        return RAMSiteInsts;
    } // end packRAMSiteInsts()

```
Now the packer correctly puts ~50% of RAMB18E1 cells in the netlist onto Sites compatible with FIFO18E1 for initial placement.

[image]

However, now I have a question about how to access Alternate SiteTypeEnums for a SiteInst, and generally what information is retained and what is lost in a SiteInst when I unPlace() it?

In the placer I have the following function that attempts to propose a new Site for a SiteInst to move to or swap with during annealing:
```java
    protected Site proposeSite(SiteInst si, List<Site> connections, boolean swapEnable) {
        SiteTypeEnum ste = null;
        SiteTypeEnum[] altStes = si.getAlternateSiteTypeEnums();
        if (altStes.length == 0)
            ste = si.getSiteTypeEnum();
        else {
            int randIndex = rand.nextInt(altStes.length);
            ste = altStes[randIndex];
            System.out.println(ste);
        }
        Site selectedSite = null;
        int attempts = 0;
        while (true) {
       // ...
```
The SiteInst si passed into the function could either be placed or unplaced (as in, intially placed, then unplaced). si.getAlternateSiteTypeEnums() here always returns a zero-length array, even when si.getSiteTypeEnum() is a SLICEL or RAMB18E1. How do I correctly get alternate SiteTypeEnums given a placed or unplaced SiteInst?

I think I can work around it by explicitly defining some logic to choose between RAMB18E1/FIFO18E1 and SLICEL/SLICEM but that seems to not be as robust.

clavin-xlnx, 2025-03-17:
> However, now I have a question about how to access Alternate SiteTypeEnums for a SiteInst, and generally what information is retained and what is lost in a SiteInst when I unPlace() it?

First, my apologies for the late response, somehow I missed the notification of your post.

The `SiteInst.getAlternateSiteTypeEnums()` method is a function of the `SiteInst`'s currently placed site.  If the `SiteInst` is unplaced, it cannot check the underlying site for alternate types that it supports since it is unplaced.  I would expect it to return a zero-length array in this case.  However, it should return a non-zero-length array in the case that the `SiteInst` is placed.

 
> In the placer I have the following function that attempts to propose a new Site for a SiteInst to move to or swap with during annealing:
> 
>     protected Site proposeSite(SiteInst si, List<Site> connections, boolean swapEnable) {
>         SiteTypeEnum ste = null;
>         SiteTypeEnum[] altStes = si.getAlternateSiteTypeEnums();
>         if (altStes.length == 0)
>             ste = si.getSiteTypeEnum();
>         else {
>             int randIndex = rand.nextInt(altStes.length);
>             ste = altStes[randIndex];
>             System.out.println(ste);
>         }
>         Site selectedSite = null;
>         int attempts = 0;
>         while (true) {
>        // ...
> The SiteInst si passed into the function could either be placed or unplaced (as in, intially placed during its creation, then unplaced). si.getAlternateSiteTypeEnums() here always returns a zero-length array, even when si.getSiteTypeEnum() is a SLICEL or RAMB18E1. How do I correctly get alternate SiteTypeEnums given a placed or unplaced SiteInst?

There is a method on the `Device` class that might be able to help here.  You'll probably want to provide it as a parameter to your `proposeSite()` method (you can get it from the `SiteInst`, but if it is not placed, it won't do any good).  Once you have a reference to the `Device` object, you can do the following:

```java
protected Site proposeSite(SiteInst si, List<Site> connections, boolean swapEnable, Device device) {
         Site[] compatibleSites = device.getAllCompatibleSites(si.getSiteTypeEnum());
         int randIndex = rand.nextInt(compatibleSites.length);
         Site proposedSite = compatibleSites[randIndex];
         // ...
```

`Device.getAllCompatibleSites()` will return all the sites in the device for which the `SiteInst` is a compatible placement destination.  This includes all the the sites that have a default `RAMB18E1` type and `FIFO18E1`.  

Let me know if you have any more questions, I'll try to keep a closer eye out for it.

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx