Xilinx/RapidWright #1183: Infinite loop for int DesignTools::updatePinsIsRouted(Net net)

coherent17, 2025-05-04:
Hi RapidWright Developer,

I'm now read_checkpoint with partially routed design in the below code snippet:
```
        design = design.readCheckpoint(dcpFile, edifFile);
        DesignTools.makePhysNetNamesConsistent(design);
        DesignTools.createMissingSitePinInsts(design);
        DesignTools.updatePinsIsRouted(design);
```

Inside design/DesignTools.java

```java
    /**
     * Update the SitePinInst.isRouted() value of all sink pins in the given
     * Design. See {@link #updatePinsIsRouted(Net)}.
     * @param design Design in which pins are to be updated.
     * @return Number of unrouted sink pins (not driven by hierarchical ports) across design.
     */
    public static int updatePinsIsRouted(Design design) {
        int totalUnroutedSinkPins = 0;
        for (Net net : design.getNets()) {
            System.out.println("Updating pins for net " + net.getName());
            int numUnroutedSinkPins = updatePinsIsRouted(net);
            if (!DesignTools.isNetDrivenByHierPort(net)) {
                totalUnroutedSinkPins += numUnroutedSinkPins;
            }
        }
        return totalUnroutedSinkPins;
    }
```

There's a net with route status [Partially routed (route conflicts)] has a really large runtime in the function ```updatePinsIsRouted(net)``` been called in ```updatePinsIsRouted(design)``` method. The queue inside the function ```updatePinsIsRouted(net)``` is unable to become empty.

```java
    public static int updatePinsIsRouted(Net net) {
        ...
        while (!queue.isEmpty()) {
            NetTools.NodeTree node = queue.poll();
            SitePinInst spi = node2spi.remove(node);
            if (spi != null) {
                spi.setRouted(true);
                if (!spi.isOutPin()) {
                    assert(numUnroutedSinkPins > 0);
                    numUnroutedSinkPins--;
                }
            }
            queue.addAll(node.fanouts);
        }
        return numUnroutedSinkPins;
    }
```

Can you suggest how to resolve this issue?

Thanks,
Coherent

eddieh-xlnx, 2025-05-05:
Hi @coherent17. Thanks for the report!

You've correctly identified that `DesignTools.updatePinsIsRouted` is not very good at handling conflicting nets -- I can imagine routing loops would cause it to fail, if not others, since it does not track which nodes it has already walked down. Is it possible to share your route so that we can be sure it gets fixed properly?

You may wish to use `CodeGenerator.testNetGenerator()` to generate some standalone Java code with the PIPs attached to the net in question.

Better yet, we'd welcome any fixes too!

eddieh-xlnx, 2025-05-05:
Also worth mentioning that -- outside of fixing this infinite loop -- if you wanted to compute the routed state of a big design more quickly, then it's absolutely possible to throw multiple threads at this analysis since it should be embarrassingly parallel and read-only!

coherent17, 2025-05-06:
> Hi [@coherent17](https://github.com/coherent17). Thanks for the report!
>
> You've correctly identified that `DesignTools.updatePinsIsRouted` is not very good at handling conflicting nets -- I can [...]

Hi @eddieh-xlnx 

Thanks for point out the problem that nodes might already walked down but keeps push into the queue.
Sorry I'm not allowed to provide the String that ```CodeGenerator.testNetGenerator(net)``` returns. 
Is it possible that I create a pull request and see if I it can pass your regression testcase?

Possible code change (available on my design matched with ```report_route_status```):

```java
    public static int updatePinsIsRouted(Net net) {
        ...

        Queue<NetTools.NodeTree> queue = new ArrayDeque<>();
        for (NetTools.NodeTree node : NetTools.getNodeTrees(net)) {
            if (node.fanouts.isEmpty()) {
                // Skip source pins with no fanout
                continue;
            }
            queue.add(node);
        }

        HashSet<NetTools.NodeTree> visited = new HashSet<>();
        while (!queue.isEmpty()) {
            NetTools.NodeTree node = queue.poll();
            if (visited.contains(node)) {
                continue;
            }
            visited.add(node);
            SitePinInst spi = node2spi.remove(node);
            if (spi != null) {
                spi.setRouted(true);
                if (!spi.isOutPin()) {
                    assert(numUnroutedSinkPins > 0);
                    numUnroutedSinkPins--;
                }
            }
            queue.addAll(node.fanouts);
        }
        return numUnroutedSinkPins;
    }
```

coherent17, 2025-05-06:
Hi @eddieh-xlnx, just create a PR #1184 on this issue, can you help verified if it is resonable fix?

Thanks

eddieh-xlnx, 2025-05-06:
Hi @coherent17. Thanks for PR#1184! While I don't believe there is anything functionally wrong with your PR, I was concerned about any performance impact that it may have so I thought about this problem a little more.

It turns out that `DesignTools.updatePinsIsRouted()` relies on `NetTools.getNodeTrees()` which already builds a map, and so I found a way to re-use that for a more general solution. Please try out #1185 to see if it fixes your original test case.

In case it's not clear, it's worth re-iterating that `updatePinsIsRouted()` will only tell you whether a pin has contiguous routing back to a source -- it does not say whether that routing is conflict-free -- if that's what you are interested in. Taking a step back: we'd be interested in what you are working on and we'd be happy to provide more specific and detailed guidance -- get in touch!

coherent17, 2025-05-07:
Hi @eddieh-xlnx. Thanks for PR #1185. I clone the branch, and it successfully run on the original testcase, can you help merge your PR into the master? Thanks for the immediate patch!

I'd like to share some of my work recently, perhaps you can suggest a better implementation method. I'm now implementing a tool that takes a DCP(might be fully routed, or partial routed due to site pin conflict) and generate a routing heat map as I discussed with @clavin-xlnx in #1144 & #1145.

However, in the PR #1145, didn't consider the partial routed net, so that the route heat map reported might be too optimistic since for some partial routed net, the PIP didn't actually occupy the INT-tile.

What I want to achieve is something that Vivado GUI mode shows "interconnect density" metric, ex: 

[image]

Below is the code snippet that calculate the routing heat map with the tile for partial routed net's PIP marked as util rate as 1.

```java
    private static void dumpRouteHeatMap() {

        // mark the unrouted nets in the tile as 1
        double [][] routeHeatMap = new double[maxRow][maxCol];

        // just calculate the PIP usage inside the INT_TILE
        double [][] switchboxUtilMap = new double[maxRow][maxCol];

        // check whether the net is routed or not routed
        ArrayList<Net> unroutedNets = new ArrayList<>();

        for (Net n : design.getNets()) {
            // skip for the nets that has no pip and the static net
            if (!n.hasPIPs() || n.getName().contains("GLOBAL_LOGIC")) continue;

            // iterate the site pins to check whether the net is routed or not
            for (SitePinInst pin : n.getPins()) {
                if (!pin.isRouted() && !pin.isOutPin()) {
                    System.out.println("Net " + n.getName() + " has unrouted pin: " + pin.getName() + " in tile: " + pin.getTile().getName());
                    unroutedNets.add(n);
                    break;
                }
            }

            // calculate the switchbox util rate by pip usage
            for (PIP p : n.getPIPs()) {
                if (p.getTile().getTileTypeEnum() == TileTypeEnum.INT) {
                    // add 2.0 to the heatmap, since for each PIP, it will be connected to 2 nodes
                    routeHeatMap[p.getTile().getRow()][p.getTile().getColumn()] += 2.0;
                    switchboxUtilMap[p.getTile().getRow()][p.getTile().getColumn()] += 2.0;
                }
            }
        }

        System.out.println("Routed Nets: " + (design.getNets().size() - unroutedNets.size()));
        System.out.println("Unrouted Nets: " + unroutedNets.size());

        // divide the value with the wirecount
        for (int i = 0; i < maxRow; i++) {
            for (int j = 0; j < maxCol; j++) {
                Tile t = tiles[i][j];
                if (!(t.getTileTypeEnum() == TileTypeEnum.INT)) continue;
                int wireCount = t.getWireCount();
                if (wireCount == 0) {
                    continue;
                } else {
                    routeHeatMap[i][j] /= (double) wireCount;
                    switchboxUtilMap[i][j] /= (double) wireCount;
                }
            }
        }

        // for unrouted nets, set the congestion value to 1
        for (Net n : unroutedNets) {
            for (PIP p : n.getPIPs()) {
                Tile t = p.getTile();
                if (t == null) continue;
                routeHeatMap[t.getRow()][t.getColumn()] = 1.0;
            }
        }

        // For each tile, if it is not an INT tile, make it share the same congestion value as the INT tile
        Set<TileTypeEnum> tileTypes = new HashSet<TileTypeEnum>();
        tileTypes.add(TileTypeEnum.CLEL_L);
        tileTypes.add(TileTypeEnum.CLEL_R);
        tileTypes.add(TileTypeEnum.CLEM);
        for (int i = 0; i < maxRow; i++) {
            for (int j = 0; j < maxCol; j++) {
                Tile t = tiles[i][j];
                TileTypeEnum tileType = t.getTileTypeEnum();

                // skip for the INT tile and the non-CLB tile
                if (tileType == TileTypeEnum.INT || !tileTypes.contains(tileType)) continue;

                Site [] sites = t.getSites();
                if (sites.length == 0) continue;

                Tile intTile = sites[0].getIntTile();
                if (intTile == null) continue;

                routeHeatMap[i][j] = routeHeatMap[intTile.getRow()][intTile.getColumn()];
                switchboxUtilMap[i][j] = switchboxUtilMap[intTile.getRow()][intTile.getColumn()];
            }
        }
        System.out.println("Route Heat Map files generated.");
    }
```

The partial visualization of the routeHeatMap is:

[image]

However, I think the result of the routeHeatMap might be too pessimistic because I mark 1 for all PIP in unrouted net. 

If there's any suggestion or bug in the above code snippet, please point out and let me know it, thanks in advance!!

eddieh-xlnx, 2025-05-07:
Since merging #1185 will close this issue, let's continue the broader discussion at https://github.com/Xilinx/RapidWright/discussions/1187.

coherent17, 2025-05-07:
Sure, thanks!!

PR #1185 (2025-05-08): [DesignTools] updatePinIsRouted() to handle routing loops
Also fixing the `NetTools.getNodeTrees()` that it depends on, by identifying nodes that are multiply-driven so that consumers (e.g. `updatePinIsRouted()`) will not repeatedly descend down routing loops.

Fixes #1183.
closed 2025-05-08 by clavin-xlnx