Xilinx/RapidWright #1186: FAILED TO ROUTE: net

lehaifeng000, 2025-05-07:
I use the bufgctrl.O to connect FDRE.C, then it will be error when routing.
my code:
```
Design design = new Design("top", "xc7a35tfgg484-2");

Net clk_ibuf = design.createNet("clk_ibuf");
Cell clk_pad = design.createAndPlaceIOB("clk",PinType.IN, "W19", "LVCMOS33", clk_ibuf, clk_ibuf.getLogicalNet());
clk_ibuf.connect(clk_pad, "O");
Cell bufg_cell = design.createAndPlaceCell("clk_IBUF_BUFG_inst", Unisim.BUFGCTRL, "BUFGCTRL_X0Y0/BUFGCTRL");
clk_ibuf.connect(bufg_cell, "I0");
bufg_cell.connectStaticSourceToPins(NetType.VCC, new String[]{"CE0", "S0"});
bufg_cell.addProperty("PRESELECT_I0", true);
// FF
Cell fdre_cell = design.createAndPlaceCell("fdre_inst", Unisim.FDRE, "SLICE_X0Y63/AFF");
fdre_cell.connectStaticSourceToPins(NetType.VCC, new String[]{"CE", "D", });
fdre_cell.connectStaticSourceToPin(NetType.GND, "R");
Net clk_ibuf_bufg = design.createNet("clk_ibuf_bufg");
clk_ibuf_bufg.connect(bufg_cell, "O");
clk_ibuf_bufg.connect(fdre_cell, "C");
fdre_cell.addProperty("IS_C_INVERTED", 0);
fdre_cell.addProperty("IS_D_INVERTED", 0);
fdre_cell.addProperty("IS_R_INVERTED", 0);
fdre_cell.addProperty("BOX_TYPE", "PRIMITIVE");
fdre_cell.addProperty("INIT", 0);

// out
Net led_obuf = design.createNet("led_obuf");
Cell led_pad = design.createAndPlaceIOB("led", PinType.OUT, "N20", "LVCMOS33", led_obuf, led_obuf.getLogicalNet());
led_obuf.connect(led_pad, "I");
led_obuf.connect(fdre_cell, "Q");

design.routeSites();

new Router(design).routeDesign();

design.writeCheckpoint("D://a.dcp");
```
what's the wrong ?

clavin-xlnx, 2025-05-09:
Sorry that you are experiencing a routing error.  The problem is that the `Router` class is not very robust and cannot route Series 7 clocks well.  Unfortunately, Series 7 devices do not have great support for clock routing.  Our main router, RWRoute, can route clocks in UltraScale+ and Versal.

lehaifeng000, 2025-05-13:
OK, thanks

eddieh-xlnx, 2025-05-13:
Closing as this appears resolved. Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by eddieh-xlnx