Xilinx/RapidWright #1192: add a cell to siteinst and fix the bel

lehaifeng000, 2025-05-15:
when I have a slice siteInst, how to place a exist cell(fdre) to the AFF bel in the siteinst ?
there is only siteinst.add(cell), I don't know how to set the bel of the cell

clavin-xlnx, 2025-05-15:
Good question!  In RapidWright, we have a logical cell (`EDIFCellInst` or its hierarchical version, `EDIFHierCellInst`) and a physical cell, `Cell`.  If you are placing a pre-existing cell that has never been placed before, most likely you'll need to create the physical cell object, `Cell`.  In this case, you would start with using the existing `EDIFCellInst` reference to create a `Cell` object using the [`Design.createCell(String instName, EDIFCellInst instance)`](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Design.html#createCell-java.lang.String-com.xilinx.rapidwright.edif.EDIFCellInst-).  Then once you've got a `Cell` object you could place it using [`Design.placeCell(Cell cell, Site site, BEL bel)`](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Design.html#placeCell-com.xilinx.rapidwright.design.Cell-com.xilinx.rapidwright.device.Site-com.xilinx.rapidwright.device.BEL-).  For example:

```java
Design d = Design.readCheckpoint("myDesignWithAnUnplacedFlipFlop.dcp");
Device device = d.getDevice();
EDIFCellInst myFlipFlop = d.getTopCell().getCellInst("myUnplacedFlipFlop"); // Assumes the cell instance already is in the netlist
Cell physicalCell = d.createCell("myUnplacedFlipFlop", myFlipFlop);
Site site = device.getSite("SLICE_X2Y2");
BEL bel = site.getBEL("AFF");
d.placeCell(physicalCell, site, bel); 
```

Currently, there is no API on the `Cell` object to place itself, it requires the `Design` in order to make sure data structures stay consistent.  

if you want to create a new flip flop that doesn't already exist in the netlist, you can use a single API that will create and place at the same time:

```java
Design d = new Design("HelloWorld",Device.PYNQ_Z1);
Cell newFlipFlop = d.createAndPlaceCell("myNewFlipFlop", Unisim.FDRE, "SLICE_X2Y2/AFF");
```

Let us know if you have more questions.

lehaifeng000, 2025-05-16:
ok, thanks

closed 2025-05-16 by lehaifeng000