Xilinx/RapidWright #1217: SiteInst.getCell(String belName) returns a cell on a different BEL

reillymck, 2025-06-12:
For some SiteInst objects, using `getCell(String belName)` to get a LUT cell, returns a FF or a CARRY4 instead of None. This happens when the LUT is being used only as a routethrough and/or constant generator. Usually the function correctly returns None. Attached is a python script with a specific SiteInst with this behavior. Opening the design checkpoint in Vivado shows no cell on the LUT BELs in the site.

[lut_cell_bug.zip](https://github.com/user-attachments/files/20717205/lut_cell_bug.zip)

Output of the python script (after rapidwright loads the design):
```
Getting A6LUT on site SLICE_X101Y140...
LUT.getName() -> Z_temp_81_reg[3]__1_i_1__0
LUT.getType() -> CARRY4
LUT.getBELName() -> A6LUT
LUT.getEDIFHierCellInst().getFullHierarchicalInstName()  -> Z_temp_81_reg[3]__1_i_1__0

Getting CARRY4 on site SLICE_X101Y140...
CARRY4.getName() -> Z_temp_81_reg[3]__1_i_1__0
CARRY4.getType() -> CARRY4
CARRY4.getBELName() -> CARRY4
CARRY4.getEDIFHierCellInst().getFullHierarchicalInstName()  -> Z_temp_81_reg[3]__1_i_1__0

Getting edif cell from netlist object with name Z_temp_81_reg[3]__1_i_1__0...
edif_cell.getName() -> Z_temp_81_reg[3]__1_i_1__0
```

clavin-xlnx, 2025-06-12:
@reillymck - Thanks for sharing your script and design.  These cells are present on purpose and represent routethru functionality.  They are actually required to properly represent the routethru to Vivado (even though they don't show up as a cell in the GUI).  

You can avoid traversing them by using the `Design.getCell(String name)` API instead of the `SiteInst.getCell(BEL bel)` one.  You can also filter them out by using `Cell.isRoutethru()`.  

We probably don't have this documented as well as we should so it is probably a TODO on our side to make this more clear in the documentation.

reillymck, 2025-06-12:
Thanks for the response. I am actually interested in finding all the LUTs that are being used as routethrus and constant generators. That is why I am iterating by site instance and querying the BEL name for the LUTs. A lot of the time it returns a LUT cell, it's just sometimes it is returning a Carry cell or a FDRE cell. These carry cells and FDRE cells do show up in the design in vivado. I understand vivado has a hidden cell to represent the routethrus and constant generators. It seems weird/invalid to return a cell for the LUTs with the same name as the CARRY4 or FDRE cell in the slice. Is that expected?

clavin-xlnx, 2025-06-13:
This is expected and required to represent the routethru correctly.

reillymck, 2025-06-13:
Thanks for the clarification!