Xilinx/RapidWright #1228: Possible missing SiteInst

reillymck, 2025-06-20:
I might be missing the proper way to do this, but I am trying to interact with this site in Rapidwright. Here is a screenshot of the site in Vivado:

[image]

However, in rapidwright there does not exist a SiteInst. `design.getSiteInst("SLICE_X4Y124")` returns None. If I try and do `design.createSiteInst("SLICE_X4Y124")`, I get the following error message:
```
Traceback (most recent call last):
  File "com.xilinx.rapidwright.design.Design.java", line 0, in com.xilinx.rapidwright.design.Design.createSiteInst
Exception: Java Exception

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<console>", line 1, in <module>
java.lang.java.lang.RuntimeException: java.lang.RuntimeException: ERROR: Failed to create site instance for site SLICE_X4Y124
```
I have a script that iterates through SiteInsts and adds routethrus and constant generators to the EDIFNetlist. It is missing this one because there is no SiteInst here, and I am lost on how to detect it.

[quad_func_dpc_edf.zip](https://github.com/user-attachments/files/20841803/quad_func_dpc_edf.zip)

reillymck, 2025-06-20:
Update, I made the incorrect assumption that site instance names were always the name of the site. I used the function getSiteInstFromSiteName, and I was able to retrieve the instance.

closed 2025-06-20 by reillymck

reillymck, 2025-06-20:
I also see that `SiteInst.getNetFromSiteWire("A6LUT_O6")` for this site returns None.
`SiteInst.getConnectedNets()` returns an empty list for this site

I do think this behavior is unexpected, since signals from LUT constant generators are present in these calls for other site_inst objects.

clavin-xlnx, 2025-06-20:
> I also see that `SiteInst.getNetFromSiteWire("A6LUT_O6")` for this site returns None. `SiteInst.getConnectedNets()` returns an empty list for this site

This is by design.  When there are not cells in a SLICE other than a static source LUT, these `SiteInst` are special and do not appear in the general netlist.  

> I do think this behavior is unexpected, since signals from LUT constant generators are present in these calls for other site_inst objects.

Unfortunately, the net to site wire routing for static sources is inconsistent in DCPs from Vivado.  Sometimes the net to site wire mappings are optional and in this scenario, I believe that is the case.