Xilinx/RapidWright #1249: Issue reproducing LDCE cell with intrasite routing

oshears, 2025-07-14:
Hi RapidWright Team,

I am having an issue reproducing a specific circuit in RapidWright that gets generated by Vivado.

### Desired Design

With Vivado and the following HDL, I am able to generate this design:

```vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

library UNISIM;
use UNISIM.VComponents.all;

entity top is
end top;

architecture Behavioral of top is
    
    signal CLE_BC_CORE_X13Y4_BNODE_TAP11_BNODE_OUTS_E19 : STD_LOGIC;
    
    attribute LOC : string;
    attribute LOC of test_SLICE_X0Y0_HFF : label is "SLICE_X0Y0";
    attribute LOC of test_SLICE_X1Y0_HLUT : label is "SLICE_X1Y0";

    attribute BEL : string;
    attribute BEL of test_SLICE_X0Y0_HFF : label is "HFF";
    attribute BEL of test_SLICE_X1Y0_HLUT : label is "HLUT";

    attribute DONT_TOUCH : string;
    attribute DONT_TOUCH of test_SLICE_X0Y0_HFF : label is "true";
    attribute DONT_TOUCH of test_SLICE_X1Y0_HLUT : label is "true";

begin

    test_SLICE_X0Y0_HFF : LDCE
        port map (
            CLR => '0',
            D   => '0',
            G   => '1',
            GE  => '1',
            Q   => CLE_BC_CORE_X13Y4_BNODE_TAP11_BNODE_OUTS_E19    
        );
    
    test_SLICE_X1Y0_HLUT : LUT1
        generic map (
            INIT => "01"          
        )
        port map (
            I0 => CLE_BC_CORE_X13Y4_BNODE_TAP11_BNODE_OUTS_E19,
            O  => open
        );

end Behavioral;
```

This gives me the following design checkpoint:

[image]

### RapidWright Design

I am attempting to do the same thing in RapidWright, but am not quite getting the same result:

```java
import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.NetType;
import com.xilinx.rapidwright.design.Unisim;

public class Test {
    public static void main(String[] args) {

        Design design = new Design("test_design", "xcku5p-ffvb676-2-e");

        Cell ldceCell = design.createAndPlaceCell("test_SLICE_X0Y0_HFF", Unisim.LDCE, "SLICE_X0Y0/HFF");
        Cell lutCell = design.createAndPlaceCell("test_SLICE_X1Y0_HLUT", Unisim.LUT1, "SLICE_X1Y0/H5LUT");

        Net connect_from = design.createNet("CLE_BC_CORE_X13Y4_BNODE_TAP11_BNODE_OUTS_E19");

        connect_from.connect(ldceCell, "Q");
        connect_from.connect(lutCell, "I0");
        
	lutCell.setSiteFixed(true);

        ldceCell.connectStaticSourceToPin(NetType.GND, "CLR");
        ldceCell.connectStaticSourceToPin(NetType.GND, "D");
        ldceCell.connectStaticSourceToPin(NetType.VCC, "G");
        ldceCell.connectStaticSourceToPin(NetType.VCC, "GE");
        ldceCell.setSiteFixed(true);

        // Perform Intra-site routing
        lutCell.getSiteInst().routeSite();
        ldceCell.getSiteInst().routeSite();

	design.writeCheckpoint("out.dcp");
    }
}
```

With this code I get the following checkpoint:

[image]

Some of the issues I am having with this design are:
1. The placed LDCE instance on the HFF BEL does not have all the pin mappings shown. The mappings for G=CLK, GE=CE and CLR=SR don't show up in Vivado, even though they appear in rapidwright when calling: `ldceCell.getPinMappingsL2P()`
2. The intra-site route created for the VCC net connecting to the G cell pin (CLK BEL pin on the HFF BEL) does not correctly configure the CLK2INV BEL in the site. When viewing the similar case for the GND net connecting to the CLR pin (SR BELPin on the HFF BEL), the RST_EFGHINV BEL does appear to be properly configured (compared to the desired design from Vivado). In this case, the global VCC net is routed from the site pin CLK2 to the input of the RST_EFGHINV BEL, and the global GND net is routed from the output of the RST_EFGHINV BEL to the SR BEL pin on the HFF BEL. For the G cell pin, I am expecting a similar behavior based on the routes from desired checkpoint routed with Vivado.

Could you all provide some guidance on how I can achieve this behavior and reproduce the desired design in RapidWright?

clavin-xlnx, 2025-07-17:
@oshears - my apologies for the late response (I was out the last week and a half and I am still trying to catch up).  

Thank you for providing the example and bringing up the issue.  Would you be willing to share the DCP that you are looking at that is generated from your RTL?  This would eliminate any potential differences if I were to generate the design.

Regarding #1 I see from the screenshots that RapidWright didn't provide the G=CLK mapping, but for GE and CLR, the results look the same (from the "Cell Properties" table in your GUI screenshots) as the results in Vivado.  In this case, the `CLK` pin is being tied to VCC which is not usually something we see.  There appears to be a minor bug in the `SiteInst.routeSite()` method that removes the pin mapping erroneously because it is attempting to emulate Vivado's behavior (`SR` and `CE` pins don't usually have pin mappings when tied to VCC).  In this case, you could add (just before `writeCheckpoint()`):
```
        ldceCell.addPinMapping("CLK", "G");
```
to match Vivado's pin mappings and work around this issue.  I have created a fix internally which will be available in the next release.

Regarding #2, I am seeing some conflicting information.  Your RTL design specifies that `G` should get `VCC` however, Vivado is configuring the site to invert VCC on the `CLK2INV` to drive GND to the `CLK` pin.  It seems that the RapidWright design matches your RTL, but Vivado does not?  If you are able to share the DCP, that would be very helpful.

oshears, 2025-07-22:
Here is the requested checkpoint Chris; And no worries, thanks for your help! 

[vivado_ldce.zip](https://github.com/user-attachments/files/21366737/vivado_ldce.zip)

oshears, 2025-07-23:
Ok, thanks for the insights! This is helpful. It has been tricky to understand how to perform some of the intrasite routing with RapidWright because of Vivado's special cases like this. I too was puzzled at CLK2INV having GND as the default output, but I think it must have something to do with the way LDCE cells are handled during implementation. I appreciate your help and guidance!