Xilinx/RapidWright #1259: Device.getBELs throws java.lang.NullPointerException

codepilot, 2025-07-29:
``` java

    public static void Unisim_to_BEL() {
        String deviceName = "xcv80";
        Device device = Device.getDevice(deviceName);

        for(Unisim u: Unisim.values()) {
            for(Entry<SiteTypeEnum, Set<String>> entry: Design.getCompatiblePlacements(device.getFamilyType(), u).entrySet()) {
                SiteTypeEnum siteTypeEnum = entry.getKey();
                try {
                    BEL[] bels = device.getBELs(siteTypeEnum);
                } catch(java.lang.NullPointerException err) {
                    System.err.println(
                        "Design.getCompatiblePlacements(FamilyType." + device.getFamilyType() + ", Unisim." + u.name() + "); " +
                        "Device.getDevice(\"" + deviceName + "\").getBELs(SiteTypeEnum." + siteTypeEnum.name() + ");" +
                        " //" + err.toString());
                }
            }
        }
    }

```

``` java

Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2PS_NOC_M_AXI); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2PS_NOC); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2PS_NOC_M_AXIS); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2PS_NOC); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2PS_NOC_S_AXI); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2PS_NOC); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2PS_NOC_S_AXIS); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2PS_NOC); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2P_PL_M_AXIS32); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2P_PL); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2P_PL_M_AXIS64); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2P_PL); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2P_PL_M_EVENTS); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2P_PL); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2P_PL_S_AXIS32); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2P_PL); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2P_PL_S_AXIS64); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2P_PL); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.AIE2P_PL_S_EVENTS); Device.getDevice("xcv80").getBELs(SiteTypeEnum.AIE2P_PL); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.BUFDIV_LEAF); Device.getDevice("xcv80").getBELs(SiteTypeEnum.BUFDIV_LEAF_ULVT); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.BUFG_FABRIC); Device.getDevice("xcv80").getBELs(SiteTypeEnum.BUFG_FABRIC_ULVT); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.BUFG_GT); Device.getDevice("xcv80").getBELs(SiteTypeEnum.BUFG_GT_ULVT); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.BUFG_HM); Device.getDevice("xcv80").getBELs(SiteTypeEnum.BUFG_PS_ULVT); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.BUFG_PS); Device.getDevice("xcv80").getBELs(SiteTypeEnum.BUFG_PS_ULVT); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.CMPHY_OCTAD); Device.getDevice("xcv80").getBELs(SiteTypeEnum.CMPHY_OCTAD); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.DDRMC5); Device.getDevice("xcv80").getBELs(SiteTypeEnum.DDRMC5); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.DDRMC5C); Device.getDevice("xcv80").getBELs(SiteTypeEnum.DDRMC5C); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.DDRMC5E); Device.getDevice("xcv80").getBELs(SiteTypeEnum.DDRMC5E); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.HARD_INV); Device.getDevice("xcv80").getBELs(SiteTypeEnum.X5PHIO_HARD_INV); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.IBUF); Device.getDevice("xcv80").getBELs(SiteTypeEnum.X5PHIO_XCVR); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.IBUFDS); Device.getDevice("xcv80").getBELs(SiteTypeEnum.X5PHIO_XCVR); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
[... cut ...]
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.X5PLL_INTF); Device.getDevice("xcv80").getBELs(SiteTypeEnum.X5PLL_INTF); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null
Design.getCompatiblePlacements(FamilyType.VERSALHBM, Unisim.X5PLL_S2P); Device.getDevice("xcv80").getBELs(SiteTypeEnum.X5PLL_S2P); //java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.j.a()" because the return value of "com.xilinx.rapidwright.device.Device.a(com.xilinx.rapidwright.device.SiteTypeEnum)" is null

```

codepilot, 2025-07-29:
checking if any compatible sites exist first makes my code not throw anymore, but the API should probably do something similar to avoid throwing

``` java

    public static void Unisim_to_BEL() {
        String deviceName = "xcv80";
        Device device = Device.getDevice(deviceName);

        for(Unisim u: Unisim.values()) {
            for(Entry<SiteTypeEnum, Set<String>> entry: Design.getCompatiblePlacements(device.getFamilyType(), u).entrySet()) {
                SiteTypeEnum siteTypeEnum = entry.getKey();
                Site compatibleSites[] = device.getAllCompatibleSites(siteTypeEnum); /*added check*/
                if(compatibleSites.length == 0) continue; /*added check*/
                try {
                    BEL[] bels = device.getBELs(siteTypeEnum);
                } catch(java.lang.NullPointerException err) {
                    System.err.println(
                        "compatibleSites: " + compatibleSites.length + "; " +
                        "Design.getCompatiblePlacements(FamilyType." + device.getFamilyType() + ", Unisim." + u.name() + "); " +
                        "Device.getDevice(\"" + deviceName + "\").getBELs(SiteTypeEnum." + siteTypeEnum.name() + ");" +
                        " //" + err.toString());
                }
            }
        }
    }

```

clavin-xlnx, 2025-07-29:
Thanks @codepilot!  I've got a fix for the exceptions being thrown in the works for a future release.  But the core of the issue is that the Versal Unisims were out of sync with the 2025.1 device models.  I've created https://github.com/Xilinx/RapidWright/pull/1261 to bring them into alignment, give it a try and it should avoid all the issues present.