Xilinx/RapidWright #1272: Issue opening and manipulating RapidWright modified DCP files in Vivado for Versal devices

oshears, 2025-08-14:
Hi RapidWright Team,

I've been running into an issue using certain RapidWright modified checkpoints in Vivado. These designs are targeting Versal devices.

After creating an initial DCP in Vivado (`pre_rapidwright.dcp`), I open the DCP in RapidWright and perform the following changes
- Insertion of new cells 
- Insertion of new nets

When I export the modified checkpoint from RapidWright (`post_rapidwright.dcp`), I get issues in Vivado when attempting the following types of commands:
- place_design
- route_design
- write_pdi

The issue is that Vivado crashes with the following type of message:

STDOUT:

```
osaze@server$vivado post_rapidwright.dcp 

****** Vivado v2024.2.2 (64-bit)
  **** SW Build 6060944 on Thu Mar 06 19:10:09 MST 2025
  **** IP Build 6050500 on Thu Mar  6 23:33:39 MST 2025
  **** SharedData Build 6060542 on Thu Mar 06 10:31:07 MST 2025
  **** Start of session at: Thu Aug 14 07:10:34 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2025 Advanced Micro Devices, Inc. All Rights Reserved.

start_gui
Abnormal program termination (11)
Please check '/nas/home/shears/Documents/cift_versal/cift-internal/code/plugins/xilinx-vivado/build/src/java/rapidwright_apps/out/hs_err_pid2306082.log' for details
segfault in /projects/cift/tools/xilinx/Vivado/2024.2/bin/unwrapped/lnx64.o/vivado -exec vivado post_rapidwright.dcp, exiting...
```

`hs_err_pid2306082.log`:

```
#
# An unexpected error has occurred (11)
#
Stack:
/projects/cift/tools/xilinx/Vivado/2024.2/tps/lnx64/jre21.0.1_12/lib//server/libjvm.so(+0xe4f55c) [0x7f0451b2f55c]
/projects/cift/tools/xilinx/Vivado/2024.2/tps/lnx64/jre21.0.1_12/lib//server/libjvm.so(JVM_handle_linux_signal+0x23e) [0x7f0451b3002e]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f048ac19520]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_netlist.so(HDLHNet::getSigType() const+0x9) [0x7f046200c459]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_constraints.so(HDPYNetProxy::HDPYNetProxy(HDLHNet const*, bool)+0x62) [0x7f0464517de2]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_constraints.so(HDPYRoutedSiteBuilder::newRoutedSitePlaceDB(HDDMSite const&)+0x3b5) [0x7f0464396855]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_constraints.so(HDPYRoutedSiteCache::find(HDPYRoutedSitePtr&, HDDMSite const&)+0x301) [0x7f0464efb4a1]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_constraints.so(PYRouteStorageAnalyzer::CheckNetlistWorker::operator()(std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > const&)+0x9f) [0x7f0463f4208f]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_constraints.so(+0x1412a34) [0x7f0463f46a34]
/projects/cift/tools/xilinx/Vivado/2024.2/lib/lnx64.o/libxv_constraints.so(std::_Function_handler<void (), xm::detail::TaskList::emplaceImpl<xm::detail::ParallelForIndividual<xm::detail::TaskList, __gnu_cxx::__normal_iterator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >*, std::vector<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >, std::allocator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > > > > >, __gnu_cxx::__normal_iterator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >*, std::vector<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >, std::allocator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > > > > >, PYParallelRun<std::map<unsigned int, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >::run<PYRouteStorageAnalyzer::CheckNetlistWorker>(PYRouteStorageAnalyzer::CheckNetlistWorker const&)::{lambda(std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > > const&)#1}>(unsigned int, __gnu_cxx::__normal_iterator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >*, std::vector<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >, std::allocator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > > > > >, __gnu_cxx::__normal_iterator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >*, std::vector<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >, std::allocator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > > > > >, PYParallelRun<std::map<unsigned int, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >::run<PYRouteStorageAnalyzer::CheckNetlistWorker>(PYRouteStorageAnalyzer::CheckNetlistWorker const&)::{lambda(std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > > const&)#1}, unsigned int)::{lambda()#1}>(xm::detail::TaskSemaphore*, xm::detail::ParallelForIndividual<xm::detail::TaskList, __gnu_cxx::__normal_iterator<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > >, std::_Rb_tree_iterator<std::pair<unsigned int const, std::vector<HDLHInstTerm const*, std::allocator<HDLHInstTerm const*> > > > >*, std::vector<std::pair<std::_Rb_tree_iterator<std::pair<unsigned int const, std: [... cut ...]
Occasionally, as with this case, I have observed that Vivado would crash just having this checkpoint opened, so I believe that the modified DCP is putting Vivado in an unstable state.

I am currently experiencing this with RapidWright 2024.2.2, but have also run into this issue with previous versions of RapidWright (2022.2.2). I've attached a tar file containing the DCP files before and after rapidwright modification.

Have you all come across any issue like this before? Do you have any suggestion for getting around this?

[dcp.tar.gz](https://github.com/user-attachments/files/21772130/dcp.tar.gz)

Thank you for your time,

Osaze Shears

oshears, 2025-08-19:
I think I am close to resolving this issue after upgrading to the newest version of rapidwright. I am still getting the internal vivado error (`hs_err_pid*.log`) for certain configurations, but I think it has to do with the way I am placing and removing cells.

With our design, we occasionally place cells and nets, then remove them if we are not able to route the design in RapidWright. I am having trouble with some things not being fully removed from the design/EDIF in RapidWright. Below I have the following code for removing nets and cells respectively:

```java
// Sample code for disconnecting ports in EDIF, assuming cells have been connected
for(EDIFHierPortInst inst : sourceCell.getEDIFHierCellInst().getHierPortInsts()){
	ECOTools.disconnectNet(design, inst);
}

// Sample code for removing nets from design/EDIF
design.removeNet(pipNet);

// Sample code for removing cells from design/EDIF
Map<Net, Set<SitePinInst>> deferredRemovals = null;
List<EDIFHierCellInst> insts = new ArrayList<>();
insts.add(cell.getEDIFHierCellInst());
ECOTools.removeCell(design, insts, deferredRemovals);     
```

Are these the recommended ways to remove Nets/Cells in RapidWright? If not, how should I be going about this process instead?

clavin-xlnx, 2025-08-20:
> Are these the recommended ways to remove Nets/Cells in RapidWright? If not, how should I be going about this process instead?

There are probably a few valid ways to accomplish what you want to do.  Without an example, it is difficult to diagnose if there could be a problem as some things are context dependent.  

RapidWright has two main views into a design, the logical netlist and the physical netlist.  The logical netlist is the hierarchical/logical view---everything you would see in an EDIF file (hence why the package is called `edif` and all classes are prefixed with `EDIF`).  The logical netlist must be in sync with the physical netlist or Vivado will likely produce an error.  The methods in `ECOTools` are designed to keep both logical and physical views in sync.  If you are using those methods and still getting errors, please share a test case with us so we can fix it.  

The physical netlist (uses classes like `Net`, `Cell`, `SiteInst`, `SitePinInst`) is essentially the structure that allows the logical netlist to be mapped to the physical device.  When operating on these objects, it is not always guaranteed that the logical netlist will be updated in sync.  For example,

```java
design.removeNet(net);
```
removes the physical net from the design such as the inter-site routing (PIPs) and intra-site routing (nets mapped to site wires) as well as removing `SitePinInst` objects connected to that `Net`.  However, it doesn't remove the corresponding logical net or any of its aliases.  There is no `ECOTools.removeNet()` method, but perhaps there should be one?  I think to accomplish that, you would need to combine `Design.removeNet()` with some number of calls to `EDIFCell.removeNet()` to remove the net logically.  

The usage for `ECOTools.removeCell()` appears correct.  So if you can share any failing examples, that would be helpful.

oshears, 2025-08-26:
Thanks for your help Chris, it seems like ECOTools and your guidance on removing the cells has resolved the internal errors I was running into. I cleaned up some of our code that was managing SitePinInsts directly and likely causing the additional errors I was getting.